rp-rs/rp-hal #525: Not able to blink the onboard led on rpi pico w

Narayanbhat166, 2022-12-16:
I'm trying to get started with running rust on my newly purchased rpi pico w ( with wifi support ). Initially when I set everything up and run the blinky example, it did not work. I then ran the micro python example and the led was working. For the w version the rpi community says that there is a slight change in the way onboard led is configured 

Link to the thread https://forums.raspberrypi.com/viewtopic.php?t=336836

Is there any change that must be made in the rust blinky example to get it working?

jannic, 2022-12-16:
"a slight change" is an understatement: The LED is connected to a different chip, the wifi one, not the RP2040.
Therefore, to blink the LED, you need a wifi driver. Which is not yet available in the released version of rp2040-hal.

If you are adventurous, you could try the experimental WIP version at https://github.com/jannic/rp-hal/tree/pico-w.
The blinky example is at https://github.com/jannic/rp-hal/blob/pico-w/boards/rp-pico-w/examples/pico_w_blinky.rs

(BTW, please don't close this ticket. There have been multiple instances of the same question in the past. Perhaps an open ticket with a descriptive title will help other people with the same issue.)

Narayanbhat166, 2022-12-17:
This works. Thanks!

metetik, 2022-12-20:
I had same issue thanks so much 🙏

dewey4iv, 2023-05-01:
Still works! :)

alexkingnz, 2023-07-17:
Is the pico-w branch going to be merged into main at some stage?  I see it's behind main by about 6 months?  Anything others could do to assist with it?

jannic, 2023-07-17:
In theory, that's the idea. In practice, it was blocked on some other crates which were not released to crates.io yet, and would probably need nightly rust.

I didn't follow recent developments closely so I'm not sure if the reasons still exist. My plan is to revisit the situation every now and then, but I don't want to spend much time on updating the branch until it is realistic to actually merge it.

cartercanedy, 2024-07-20:
Is there any progress on pico w support?

xFranv8, 2025-02-13:
Any progress on this?

9names, 2025-02-13:
There has been no progress on this.
embassy-rp has good Pico W support, you should use that if you have a Pico W or Pico 2W