rp-rs/rp-hal #528: Rename DMA SingleBuffering, Bidirectional?

9names, 2022-12-19:
Another suggestion: Could you rename `SingleBuffering` to `SingleBufferedTransfer` (and `Bidirectional` accordingly)?
Found the other names a little clumsy when trying it out.

_Originally posted by @jounathaen in https://github.com/rp-rs/rp-hal/issues/457#issuecomment-1339176354_

9names, 2022-12-19:
Should we rename this Structs? What should we name them to?
Should we keep the old ones around (deprecated) for one release to give people who used the old DMA code some time to migrate?

jounathaen, 2022-12-19:
I think that there is no need to take people using code from the unmerged PRs into consideration. I'm also affected by this, but when using such code, it is clear that it is subject to breaking changes anytime. People using the old code can still use the respective git commit for the HAL crate until they migrate.

jannic, 2022-12-19:
I do not yet have an opinion about the name. I can see the point of renaming it, because `SingleBuffering` sounds incomplete without a noun. 
If, in common usage, structs of that type are only used in contexts where it's obvious that it's a kind of transfer, that would be fine. As I didn't use this API yet, I don't know how common usage of those structs would look like.

Regarding deprecation, I agree with @jounathaen: There's no need for keeping a deprecated version around for code that has never been merged to the main branch.

9names, 2022-12-19:
@jounathaen how do you feel about:
SingleBuffering becomes Transfer
DoubleBuffering becomes DoubleBufferedTransfer
Bidirectional becomes BidirectionalTransfer

What about the namespaces? Okay as-is, or should they be updated too?

jounathaen, 2022-12-19:
>  SingleBuffering becomes Transfer
DoubleBuffering becomes DoubleBufferedTransfer
Bidirectional becomes BidirectionalTransfer

I like that! :+1: 

> What about the namespaces? Okay as-is, or should they be updated too?

Hmmm. Not exactly sure about that. My thoughts:

- It is generally speaking okay as is. Has a little clumsiness to it, but not too much.
- Putting at least the transfers in the parent namespace might be nice 
  - `dma::Transfer` vs `dma::SingleBuffering::Transfer`
  -  `dma::DoubleBufferedTransfer` vs `dma::DoubleBuffering::DoubleBufferedTransfer`

Alternative Idea: Name everything `Transfer`/`Config` and distinguish by namespace:
-  `dma::SingleBuffering::Transfer`
-  `dma::SingleBuffering::Config`
-  `dma::DoubleBuffering::Transfer`
-  `dma::DoubleBuffering::Config`
- ...

9names, 2022-12-19:
> Putting at least the transfers in the parent namespace might be nice 

We already `pub use` these at the `dma` level, so after renaming it would already be exposed as `dma::Transfer`, `dma::TransferConfig`,  `dma::DoubleBufferedTransfer`, etc.

> Alternative Idea: Name everything Transfer/Config and distinguish by namespace:

This would be consistent and have a less cluttered top-level namespace, but it does mean users have to explicitly disambiguate or alias these.
If we go this way I'd want to go with slightly less verbose namespaces:
`dma::SingleBuffer::Transfer`
or maybe
`dma::DoubleBuf::Transfer`

jounathaen, 2022-12-19:
> This would be consistent and have a less cluttered top-level namespace, but it does mean users have to explicitly disambiguate or alias these.

True. That's the drawback. But isn't that's what namespaces are for in the end. :thinking: 

> ` dma::SingleBuffer::Transfer`

I think that sounds nice.

closed 2022-12-20 by 9names