rp-rs/rp-hal #536: SPI slave implementation not available

DmitriLyalikov, 2022-12-30:
I am not sure that the functionality of configuring the peripheral as a slave exists yet, and I think this is an important feature. If so,
please let me know if this can be done with my own pull request, or if this will be implemented later.

jannic, 2022-12-30:
You are right, that feature is still missing. And contributions are welcome!
There is some information about how to contribute in the [README](https://github.com/rp-rs/rp-hal#contributing).

zaksabeast, 2023-01-03:
@DmitriLyalikov are you working on this?  If not, I'd be happy to submit a PR.

DmitriLyalikov, 2023-01-08:
@zaksabeast Hi, i have not gotten to this yet, go ahead with submitting and we can work from there.

PR #537 (2023-02-03): Add SPI slave
Closes #536.
closed 2023-02-03 by jannic