rp-rs/rp-hal #554: Configure TransmittedWord for PIO SM DMA

chrisprice, 2023-02-23:
It is not possible to configure DMA to use a 1 byte transfer size when writing to the PIO SM. It is currently [hard-coded to u32](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/src/pio.rs#L1583) -

```rust
impl<SM: ValidStateMachine> WriteTarget for Tx<SM> {
    type TransmittedWord = u32;
    // ...
}
```

I believe this is possible based on section 2.5. DMA section of the datasheet -

> The transfer size can be either 32, 16, or 8 bits. This is configured once per channel: source transfer size and
destination transfer size are the same. The DMA performs standard byte lane replication on narrow writes, so byte data
is available in all 4 bytes of the databus, and halfword data in both halfwords.

I'm happy to make the change but would like some advice on where to start. 

I can see that SPI has a similar-ish thing going on and there's a [generic to support it](https://github.com/rp-rs/rp-hal/blob/636992e311bdd0a457e5797208a1c0b888756208/rp2040-hal/src/spi.rs#L422). Would a new generic parameter representing expected word size on [Tx](https://github.com/rp-rs/rp-hal/blob/636992e311bdd0a457e5797208a1c0b888756208/rp2040-hal/src/pio.rs#L1412) and associated field (Defaults to `u32`. Named `write_target_word_size`?) on the [PIOBuilder](https://github.com/rp-rs/rp-hal/blob/636992e311bdd0a457e5797208a1c0b888756208/rp2040-hal/src/pio.rs#L1840) be appropriate?

Whilst I don't have a use-case for it myself, if I was adding the above would it also be appropriate to add another to allow control of read target word size?

### Context 

This is to avoid pre-procesing the data (e.g. [as performed here allow for left shifting](https://github.com/rp-rs/ws2812-pio-rs/blob/main/src/lib.rs#L175)). I plan on combining this with a [change in the rgb crate](https://github.com/kornelski/rust-rgb/commit/fb667ccdbd481bbd589a812018146d52beed783c) to allow for direct DMA transfers.

ithinuel, 2023-02-26:
For what it's worth, I'm have a similar issue with implementing DMA support for SPI-PIO.

I'm adding this limitation to my backlog of things to work on :)

jannic, 2023-05-29:
I added this to milestone 0.9.0 just in case it needs a breaking change to fix this.

jannic, 2024-02-22:
@ithinuel:
 
> I'm adding this limitation to my backlog of things to work on :)

Is there anything in the works for 0.10.0, or should we move this ticket to the next milestone?

ithinuel, 2024-02-24:
It will need to move to the next milestone.

PR #788 (2024-04-06): Enable transfer size of PIO DMA to be specified
Rather than always assuming PIO DMA transfers are u32, also allow u16 and u8. The hardware will replicate sub-32-bit transfers across the whole 32 bit word, which is the same as accessing the Tx/Rx fifo directly.

This implements a rather general type-level description of a transfer, so maybe it should go in the `dma` module rather than being `pio` specific?

This fixes #554
commit 9947302a (2024-04-06): Enable transfer size of PIO DMA to be specified

Rather than always assuming PIO DMA transfers are u32, also allow u16
and u8. The hardware will replicate sub-32-bit transfers across the
whole 32 bit word, which is the same as accessing the Tx/Rx fifo
directly.

This implements a rather general type-level description of a transfer,
so maybe it should go in the `dma` module rather than being `pio`
specific?

This fixes #554
closed 2024-04-06 by jannic