rp-rs/rp-hal #563: Substract with overflow not detected at build time ?

pmirabel, 2023-03-06:
Hello,
Pretty new to rust embedded, I was working on a game-of-life automata as a starter project.  
Thank you all that contribute to port async rust to embedded world  :)   

I found earlier today, that I can upload a code which cause panick at runtime :  
```rust
 fn count_alive_neighbors(&self, x: usize, y: usize) -> u8 {
        // ... //
        for dy in 0..=2 {        // dy: usize
            for dx  in 0..=2 {   // dy: usize
                // ... //
                let nx = x + dx - 1; // This line cause ERROR panicked at 'attempt to subtract with overflow' if dx==0
                // ... //
            }
        }
        count
    }
```

Neither `cargo clippy`nor `cargo check` will tell me that ! Do you think it's the normal behavior or do I miss something here ?

Have a good day :)

jannic, 2023-03-06:
Hi @pmirabel!
I fear that's just how rust works, and in no way specific to rp-rs:
https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=69966375cac6501c8e9a26f4846def33

In short, while rustc _may_ detect under/overflows at compile time, it can't guarantee that. See this code for comparison:
https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=cb5c6252c6b369d6a5a58f0f3f00a11e

Here, rustc obviously knows at compile time that the code will underflow, and gives an appropriate error.
But in your example, depending on the value of `x`, an underflow may or may not happen. The compiler can't know. And even if the function is always called with `x == 0`, like in the the first playground example above, the compiler doesn't do this analysis over function boundaries.

And it gets worse: The panic isn't guaranteed either. If you compile in release mode, which is often the case for embedded software, the code is allowed to silently over/underflow. If you run the first playground example in release mode, it just prints `18446744073709551615`!

In short: While rustc is great at catching bugs at compile time, it doesn't reliably prevent overflows, so be cautious!

pmirabel, 2023-03-06:
Thank you very much ! 
  
There should be a button "close with comment and some tears" ;)

closed 2023-03-06 by pmirabel