rp-rs/rp-hal #564: Provide functions to safely halt or reset the RP2040

jannic, 2023-03-06:
A discussion on https://matrix.to/#/#rust-embedded:matrix.org showed that it's difficult to safely halt or reset a multicore MCU.

It would be nice if the HAL provided simple to use methods to halt / reset the full chip. Those could also be used from panic / HardFault handlers.

For reset, this seems to work:
```
        (*pac::PSM::PTR).wdsel.write(|w| w.bits(0x0000ffff));    
        (*pac::WATCHDOG::PTR).ctrl.write(|w| w.trigger().set_bit());    
```
However, it has the disadvantage that it also breaks the debugger connection.
Alternatively, this will first turn core1 off and then reset core0:
```
        (*pac::PSM::PTR).frce_off.write(|w| w.proc1().set_bit());                                                                                                                                               
        pac::SCB::sys_reset();                                                                                                                                                                                  
```
This doesn't break the debugger connection. But it won't work when called from core1.

Similarly, for halting the MCU from core0, this works:
```
        (*pac::PSM::PTR).frce_off.write(|w| w.proc1().set_bit());                                                                                                                                               
        loop {}
```
Some method that works from both cores the same way would be nicer.

thejpster, 2023-03-07:
Is there a register we can read to know which core we are? Then we can do the right thing no matter.

9names, 2023-03-07:
Yes, the sio module has a register for this:
https://docs.rs/rp2040/latest/rp2040/sio/cpuid/index.html

closed 2023-09-08 by jannic