rp-rs/rp-hal #565: No way to set a StateMachine's execctrl

Gip-Gip, 2023-03-07:
I am porting over an SDIO driver and for the code to work the status register needs to be set when there are 2 words in the fifo buffer. Currently, as far as I'm aware, there is no way to set the state machine's execctrl using what is currently provided.

With the PIOBuilder struct there should be a way to set the register like [here](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/e9040c36340210cd1777a8f5b3e12ef6f838c3f9/src/sd_driver/SDIO/rp2040_sdio.c#L858)

The code I'm trying to port:
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/src/sd_driver/SDIO/rp2040_sdio.c
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/src/sd_driver/SDIO/rp2040_sdio.pio

Gip-Gip, 2023-03-07:
Looks like the function to set mov_status was left out accidentally in the PIOBuilder function, just added it and tested it with the local crate and it works perfectly.

Already made a pr

jannic, 2023-03-09:
Fixed by #566

closed 2023-03-09 by jannic