rp-rs/rp-hal #572: Suggestion: Re-export foreign types from extern crates in public APIs

KizzyCode, 2023-03-26:
## Problem
Consider re-exporting foreign types in public APIs. For example, in https://github.com/rp-rs/rp-hal/blob/715c3c9dd85f4b30d949c310b60e7621ea62c3f6/rp2040-hal/src/timer.rs#L198 there is `MicrosDurationU32` from `fugit`.

The "problem" is: To use this API, I also have to add a compatible version `fugit` to my crate manually – which is obviously possible, but feels redundant because it splits my dependency tree.
```
Clean dependency tree: my_crate ─ rp2040_hal ─ fugit

Current dependency tree: my_crate ─ rp2040_hal ─ fugit
                                  ╲
                                    fugit
```
This might become especially cumbersome if my crate – for some reason – have to use another, possibly API-incompatible version of fugit.

## Suggestion
A possible alternative would be to publicly re-export the used type, so that I could simply `use rp2040_hal::timer::{Alarm, Alarm0, MicrosDurationU32}` without having to synchronize my crate's imports.

jannic, 2023-03-27:
Sounds like a reasonable request. As it's not specific to this crate: Are there any general best practices covering this question?

jannic, 2023-03-27:
Here is an old discussion about this question:
https://github.com/rust-lang/api-guidelines/discussions/176
No clear conclusion, but at least they don't say it's a bad idea to do the re-export in general.

jannic, 2023-03-27:
Another open question is where to put the re-export.
For your example it would be in `rp2040-hal::timer`, so one can do `use rp2040_hal::timer::{Alarm, Alarm0, MicrosDurationU32}`.
But fugit is also used in several other modules. So it might be better to put it at the top level?
Like this?
```
use rp2040_hal::MicrosDurationU32;
use rp2040_hal::timer::{Alarm, Alarm0};
```
Or perhaps like this?
```
use rp2040_hal::fugit::MicrosDurationU32;
use rp2040_hal::timer::{Alarm, Alarm0};
```

jannic, 2023-03-27:
I implemented the last suggestion in #573, but that's open for discussion. No need for extended bike-shedding, but if somebody knows some good reason why the re-export should be somewhere else (or why it's a bad idea in the first place), please speak up.

KizzyCode, 2023-03-27:
I also think that if fugit is used in multiple places, it's best to simply re-export the entire crate as sub-dependency (as done in the PR) instead of re-exporting individual types somewhere within `rp2040_hal`.

PR #573 (2023-10-30): Re-export fugit
Closes #572

I also had a quick look for other crates to re-export:

One candidate would be `embedded-hal`, but I think a re-export is not appropriate there: That crate is, in general, not used to access features of `rp2040-hal`, but it's the other way around: Some driver expects a type implementing a trait from `embedded-hal`, and `rp2040-hal` provides that type. In that chase, the needed version of `embedded-hal` is selected by the driver, not by `rp2040-hal`, and a re-export would not help with that.

There are probably more, but it needs to be checked manually which depende [... cut]
closed 2023-10-30 by jannic