rp-rs/rp-hal #581: Current defmt version is broken

jannic, 2023-04-12:
If you run into the following error, it is caused by a breaking change of the `defmt` crate in version 0.3.5:
```
Error: defmt wire format version mismatch: firmware is using 4, `probe-run` supports 3
suggestion: `cargo install` a different version of `probe-run` that supports defmt 4
```
Unfortunately, the latest version of `probe-run` is also broken, an update would cause this error:
```
Error: unable to determine reset handler
```
So for now, the best workaround is to stay with older versions of `probe-run` and `defmt`.
Run:
```
cargo install probe-run --version=0.3.6
```
And add this to your `Cargo.toml`
```
defmt = "=0.3.2"
```
(Version 0.3.3 would be better if it wasn't yanked for some reason. :disappointed:)

jannic, 2023-04-12:
Possible fixes for the mentioned issues:
- https://github.com/knurling-rs/probe-run/pull/398
- https://github.com/knurling-rs/defmt/pull/750

Ragarnoy, 2023-04-12:
do any of the two prs fix the issue?

jannic, 2023-04-12:
Those are two different issues, both with their own fix. Depending on what you want to do, one might be sufficient. Eg. if you update your firmware to use the most recent defmt version, you don't need support for the older one. So the fix for the reset handler issue in probe-run would be sufficient.

9names, 2023-04-13:
> cargo install probe-run --version=0.3.6

Since we've already had unintended dependency breakage, best make it
```system
cargo install probe-run --version=0.3.6 --locked
```
to ensure that nothing else new and broken gets pulled in

jannic, 2023-04-13:
Good point. It doesn't matter for the dependency in question, as probe-run does depend on an exact version of defmt-decoder. But of course a similar breakage could happen with some other dependency. I added the flag.

elpiel, 2023-04-19:
If anyone wants to help and test the Handle reset vector PR in order to be approved and merged faster, please consider commenting in the PR after testing with the changes:

```
cargo install probe-run --git https://github.com/knurling-rs/probe-run.git --rev 884e799dd7fb0cf6d64619a23a3807ca2ccb47e9
```

jannic, 2023-05-14:
Fixes for both mentioned issues have been released.

closed 2023-05-14 by jannic