rp-rs/rp-hal #591: rp2040_hal I2CPeripheralEventIterator usage

np33kf, 2023-05-03:
Hi,

Newcomer in Rust.

Coming from C# and the .NET world, I needed to go bare metal with some IC (in this case the rp2040), and choose Rust, and been loving it. And this HAL has been amazing, making the task very easy.

Until now, I been able to easily use gpio, I2C as master, but my usage case, the rp2040 will be acting as a slave (or peripheral as the HAL calls it).

I'm able to see the device address when scanning the bus with Linux i2cdetect, but I am struggling with reading and writing to the device.

Sadly i wasn't able to see any useful example/usage case online, I'm wandering if someone has some suggestion on how to proceed.

Also relevant, I am not an expert in I2C, so bear with me :)

My main loop looks like this:

``` match i2c.next() {
            Some(I2CEvent::TransferRead) => {
                info!("transfer read");
                let write: [u8; 1] = [0x55];
                let byteswrite = i2c.write(&write);
                info!("{}", byteswrite);
            }
            Some(I2CEvent::TransferWrite) => {
                info!("transfer write");
                let mut result: [u8; 1] = [0; 1];
                let mut bytesread: usize;
                loop {
                    bytesread = i2c.read(&mut result);
                    if bytesread == 0 {
                        break;
                    }
                    info!("{:#04x}", &result);
                }
            }
            Some(I2CEvent::Start) => {}
            Some(I2CEvent::Stop) => {}
            Some(I2CEvent::Restart) => {
                info!("Restart");
            }
            None => {}
        }```

I see activity with the events been triggered, but I until now, not able to read and write with Linux i2cget and i2ctransfer from another device.

Thanks in advance for any help!

Ricardo

ithinuel, 2023-05-04:
Hello,
I am using it from [here](https://github.com/ithinuel/wilskeeb/blob/main/firmware/src/inter_board/secondary.rs) and from the snippet you shared it seems correct to me.

np33kf, 2023-05-04:
Hello,

Thanks for the reply.

The snippet showed above is working. Today when using the code again it was working without a change... 

I think it was some process in Linux that was using the bus, allowing to discover the device but not read/write.

Also breakpoints in the rust code, didn't help...

Thanks for the reply and also the reference that you sent me. It's going to be useful to get deeper in rust!

closed 2023-05-04 by np33kf