rp-rs/rp-hal #592: Make `bad::Never` public

KizzyCode, 2023-05-04:
Please consider to expose [`bad::Never` from `multicore.rs`](https://github.com/rp-rs/rp-hal/blob/c6c563fc1ca261f404e2e9e4fda924f55023a95b/rp2040-hal/src/multicore.rs#L260).

## Reasoning
This type is used in a public function signature ([`Core::spawn`](https://github.com/rp-rs/rp-hal/blob/c6c563fc1ca261f404e2e9e4fda924f55023a95b/rp2040-hal/src/multicore.rs#L147)). If the function is called directly, this is not a problem. However if I – for some reason – want to write a wrapper around `Core::spawn`, my wrapper cannot enforce the same trait restrictions since the type is private and thus cannot call `Core::spawn`.

There is a workaround (reimplement `bad::Never` in your own crate), but that's not very pretty IMO 😅

ithinuel, 2023-05-04:
Since https://crates.io/crates/bad supports `no_std` couldn't we use it directly ? (and extend its use to all the `type Error` are are infallible?)

jannic, 2023-05-04:
Opposite suggestion: Assuming that there are good reasons why the never type is not yet stable, shouldn't we try to avoid using this hack at all? The spawned function could return () instead, and the spawn method could loop after calling it, for example.

ithinuel, 2023-05-04:
Haha @jannic I did this alternative PR before seen your comment :)

There's a little question left as a TODO in the PR.
When the closure returns we recover the responsibility for the interrupts etc. Should we try devise something to disable them (not sure it's possible to enforce anything reliably there), or document that the closure MUST return with the system back into a neutral state (nvic and other core specific blocks to their reset state) ?

jannic, 2023-05-04:
Also see https://github.com/rp-rs/rp-hal/pull/593#issuecomment-1534234377

(This discussion is distributed over too many tickets)

commit 4ab43cbf (2023-05-20): Merge pull request #594 from ithinuel/do-not-require-multicore-callback-to-never-return

Remove the requirement for multicore tasks to never return

Closes #592
closed 2023-05-20 by jannic