rp-rs/rp-hal #599: Fix / document on-target tests

jannic, 2023-05-05:
Some of the on-target-tests fail:
```
$ ./run_tests.sh
[...]
error: 2 targets failed:
    `--test dma_drop`
    `--test dma_m2m_u64`
```
I have no idea how dma_drop should work, and as I understand it, dma_m2m_u64 should be expected to fail, as the rp2040 DMA does only do 32bit transfers.

And then, the loopback tests need a connection from GPIO4 to GPIO7. That should be documented.

9names, 2023-05-06:
Sorry, should have documented these better, especially wiring for the loopback tests and the fact that drop doesn't work with the current implementation.

The u64 test should fail, and the test itself is decorated to expect a fail as per the defmt-test docs - did I do it wrong?
https://github.com/rp-rs/rp-hal/blob/cd4cb7baea9db9fa116fa81a3737277f56e5a987/on-target-tests/tests/dma_m2m_u64.rs#L80

ithinuel, 2023-05-06:
It fails with:

```
(1/1) running `dma_mem2mem_u64`...
└─ dma_m2m_u64::tests::__defmt_test_entry @ tests/dma_m2m_u64.rs:81
ERROR panicked at 'DMA does not support transferring 64bit data', /home/ithinuel/Documents/rust/wilskeeb/vendor/rp-hal/rp2040-hal/src/dma/single_channel.rs:149:9
└─ panic_probe::print_defmt::print @ /home/ithinuel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:181:5
   2: __udf
        at ./asm/lib.rs:51:17
   3: cortex_m::asm::udf
        at /home/ithinuel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/asm.rs:43:5
   4: panic_probe::hard_fault
        at /home/ithinuel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:86:5
   5: rust_begin_unwind
        at /home/ithinuel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:54:9
   6: core::panicking::panic_fmt
        at /rustc/9ecda8de85ce893cc3fc748ab06be0b8250147a7/library/core/src/panicking.rs:67:14
   7: core::panicking::panic
        at /rustc/9ecda8de85ce893cc3fc748ab06be0b8250147a7/library/core/src/panicking.rs:117:5
   8: main
        at tests/dma_m2m_u64.rs:37:1
   9: Reset
(HOST) ERROR the program panicked
error: test failed, to rerun pass `--test dma_m2m_u64`
```

jannic, 2023-05-06:
`should_error` doesn't support panics, as explained in the ticket introducing the feature: https://github.com/knurling-rs/defmt/issues/375

ithinuel, 2023-05-06:
Indeed I just check the source (of truth :D)
The outcome (return type of the test function) can either be unit `()` or a `Result<_, _>`.

9names, 2023-05-06:
> should_error doesn't support panics as explained in the ticket introducing the feature: https://github.com/knurling-rs/defmt/issues/375
Ah. Missed that part.

jannic, 2023-05-06:
#600 makes the 64bit transfer test unnecessary.

PR #601 (2023-05-06): Fix / document on-target tests
Closes #599
closed 2023-05-06 by ithinuel