rp-rs/rp-hal #605: How to use different pins when using spi?

miam-miam100, 2023-05-07:
I have been following https://github.com/rp-rs/rp-hal-boards/blob/main/boards/rp-pico/examples/pico_spi_sd_card.rs to try and get SD Card reading working on my PI pico W with a cytron maker pi pico board.

The board uses the following pins to interface with the SD Card:
[image]

however when I change the example code from this:
```rust
    let _spi_sclk = pins.gpio2.into_mode::<gpio::FunctionSpi>();
    let _spi_mosi = pins.gpio3.into_mode::<gpio::FunctionSpi>();
    let _spi_miso = pins.gpio4.into_mode::<gpio::FunctionSpi>();
    let spi_cs = pins.gpio5.into_push_pull_output();

    // Create an SPI driver instance for the SPI0 device
    let spi = spi::Spi::<_, _, 8>::new(pac.SPI0);
```
to this: 
```rust
    let _spi_sclk = pins.gpio10.into_mode::<gpio::FunctionSpi>();
    let _spi_mosi = pins.gpio11.into_mode::<gpio::FunctionSpi>();
    let _spi_miso = pins.gpio12.into_mode::<gpio::FunctionSpi>();
    pins.gpio13.into_pull_up_disabled();
    pins.gpio14.into_pull_up_disabled();
    let spi_cs = pins.gpio15.into_push_pull_output();

    // Create an SPI driver instance for the SPI1 device
    let spi = spi::Spi::<_, _, 8>::new(pac.SPI1);
```
I get a `TimeoutWaitNotBusy` error. How am I meant to correctly change the spi pins used so I can interface with my sd card?

ithinuel, 2023-05-07:
As far as I can see on the [schematic](https://cdn.shopify.com/s/files/1/0176/3274/files/MAKER-PI-PICO_v1.2.0_Schematic.pdf?v=1617963795) there is no pull-up on the data/clock/cs lines.
Did you add them on your board?

The current GPIO API does not allow to configure the pull up/down/both when using the spi function (or any other non-sio function for that matter).
There are upcoming changes in https://github.com/rp-rs/rp-hal/pull/585 that should help resolving this.

miam-miam100, 2023-05-07:
Ah I see thanks for your help! I've tried patching to use your experimental branch as I don't have any resistors with me but it still doesn't work. Is your PR meant to be working for spi stuff or have you not yet gotten to that yet?

ithinuel, 2023-05-08:
I haven't got to try it yet on the hardware but i happen to have it setup so I'll try that tomorrow 👍

miam-miam100, 2023-05-08:
Changed the baudrate and now everything works :smiley:

closed 2023-05-08 by miam-miam100

ithinuel, 2023-05-08:
Just an FYI: tested the GPIO overhaul and enabling the internal pull up is enough :)

They are rather strong resistors (so weak pull-ups) so this may be limiting the possible speed.