rp-rs/rp-hal #615: Generic inter-core FIFO

fla-t, 2023-05-18:
This is more like a discussion than an issue, Read the code for `Sio` and specifically `SioFifo`. 

I was trying to find an inter-core fifo example where instead of a `u32` type there is sort of a generic or bytes that I can send and pull. Then I noticed that the [SioFifo::write(&mut self, value: u32)](https://github.com/rp-rs/rp-hal/blob/8b37ffe3d5003c710a2417c2628b9ed942fe78db/rp2040-hal/src/sio.rs#LL126C2-L126C2) only writes bytes to a memory location, and instead of using a generic type the author has gone for a `u32`. Is there a reason for this `u32` usage instead of a generic one?

If it's doable, I am more than willing to contribute and make an infra that supports generics.

jannic, 2023-05-19:
It may be better for both usability and performance to use the SioFifo only for coordination purposes, and do the actual data transfer by passing pointers to memory locations. But I'm not aware of any actual implementations. So having some practical examples would be very welcome!

fla-t, 2023-05-19:
I am working on giving some examples, I don't think the implementation would differ from the one we have now but don't take my word for it I am a novice in Rust.

I will share here If I managed to pull up something, hopefully this weekend.

fla-t, 2023-05-27:
The Fifo is based on a single register? couldn't we just have used a small portion of the stack for this?