rp-rs/rp-hal #630: rp_pico::rp2040_pac::i2c0::RegisterBlock needs to be a smart pointer

RayLucchesi, 2023-06-02:
newbie here so bear with me. I'm trying to operate the I2C bus while inside a timer_irq_0. In order to do this, I want to pass the pointer to the i2c0 (which is a RegisterBlock) to the interrupt in a mutex controlled static. such as this:

static mut LED_AND_ALARM_AND_DAC: Mutex<RefCell<Option<LedAndAlarmAndDac>>> = Mutex::new(RefCell::new(None)); 

Where LedAndAlarmAndDac is defined as follows:
type LedAndAlarmAndDac = (
    hal::gpio::Pin<hal::gpio::bank0::Gpio25, hal::gpio::PushPullOutput>,
    hal::timer::Alarm0,
    mcp4725::MCP4725<hal::I2C<pac::i2c0::RegisterBlock, (hal::gpio::Pin<Gpio26, hal::gpio::Function<hal::gpio::I2C>>, hal::gpio::Pin<Gpio27, hal::gpio::Function<hal::gpio::I2C>>)>>,
);

When I try to compile the code I get:

static mut LED_AND_ALARM_AND_DAC: Mutex<RefCell<Option<LedAndAlarmAndDac>>> = Mutex::new(RefCell::new(None)); // c21.out & c25.out
    |                                                         ^^^^^^^^^^^^^^^^^ the trait `Deref` is not implemented for `rp_pico::rp2040_pac::i2c0::RegisterBlock`
    |
    = help: the trait `_embedded_hal_blocking_i2c_Read` is implemented for `rp2040_hal::I2C<T, PINS>`
    = note: required for `I2C<RegisterBlock, (Pin<Gpio26, Function<I2C>>, Pin<Gpio27, Function<I2C>>)>` to implement `_embedded_hal_blocking_i2c_Read`
    = note: the full type name has been written to '/home/pi/rp-hal-boards/target/thumbv6m-none-eabi/debug/examples/b_blinky-8a6ca0e44c025344.long-type-14025323175658563897.txt'
note: required by a bound in `mcp4725::MCP4725`
   --> /home/pi/.cargo/registry/src/github.com-1285ae84e5963aae/mcp4725-0.4.2/src/lib.rs:60:10

and this is what the compiler long_type-14025323175658563897.txt shows:

rp2040_hal::I2C<rp_pico::rp2040_pac::i2c0::RegisterBlock, (rp2040_hal::gpio::Pin<Gpio26, rp2040_hal::gpio::Function<rp2040_hal::gpio::I2C>>, rp2040_hal::gpio::Pin<Gpio27, rp2040_hal::gpio::Function<rp2040_hal::gpio::I2C>>)>

Which seems just about what I was trying to do without the mcp4725 crate type prefix.

I could possible update the a local crate with a simple Deref function for RegisterBlock but it would seem like most IRQ handlers that want to have access to I2c and other RegisterBlocks would have the same problem. So I'm thinking that maybe this should be fixed in rp2040_pac.

Sorry if it turns out I'm confused about how to declare a mutex static for passing into the IRQ, just a newbie here. 
[b_blinky_c25_rs.txt](https://github.com/rp-rs/rp-hal/files/11639525/b_blinky_c25_rs.txt)

I'm just amending the example in rp-hal/rp2040-hal/examples/vector_table.rs. I don't have any need to change the IRQ handler address but I would like to send info over the I2c to the MCP4725 DAC and change the timer alarm value. 

I've attached the whole source code as a txt file hope that helps.
R

RayLucchesi, 2023-06-02:
Not sure why the whole static definition was not shown above but here it is:

static mut LED_AND_ALARM_AND_DAC: Mutex<RefCell<Option<LedAndAlarmAndDac>>> = Mutex::new(RefCell::new(None)); // c21.out & c25.out

again. Note that the Option<...> now refers to the type I'm trying to pass. This is the way it was in the source code. Not sure what I copied above.

sorry for any confusion.

RayLucchesi, 2023-06-02:
The static is not showing properly in this issue it should be:

"static mut LED_AND_ALARM_AND_DAC: Mutex<RefCell<Option<LedAndAlarmAndDac>>> = Mutex::new(RefCell::new(None)); // c21.out & c25.out" 

I placed it in quotes here hoping that would help. 3rd times a charm...

RayLucchesi, 2023-06-02:
But no that didn't work either Trust me 

it looks like .... Option<LedAndAlarmAndDac>>> = Mutex::new(RefCell::new(None)); // c21.out & c25.out

4th time trying this. I have summarized the prefix with ... and just shown what's on the statement line from Option onwards.

RayLucchesi, 2023-06-02:
wth? I guess you will need to trust me on this inside the Option<...> chevrons,
 is the type LedAndAlarmAndDac

9names, 2023-06-02:
If you put your code between triple backticks it won't get interpreted as markdown directives.
Eg

```
By putting 3 ` at the beginning and end, github will display this as is: <Magic>
```

ithinuel, 2023-06-11:
Hi, I have a hard time reading through the misformatted message but a thing that seems odd to me is the first type in:

```rust
rp2040_hal::I2C<rp_pico::rp2040_pac::i2c0::RegisterBlock, (rp2040_hal::gpio::Pin<Gpio26, rp2040_hal::gpio::Function<rp2040_hal::gpio::I2C>>, rp2040_hal::gpio::Pin<Gpio27, rp2040_hal::gpio::Function<rp2040_hal::gpio::I2C>>)>
```
It should not be the `RegisterBlock` type but either `rp2040_pac::I2C0` or `rp2040_pac::I2C1` which both implement `Deref<Target = rp2040_pac::i2c0::RegisterBlock>`.

jannic, 2024-06-13:
@RayLucchesi: Is your question answered?