rp-rs/rp-hal #634: Release 0.9.0?

oOMelon, 2023-06-17:
Hi, is it possible to release version 0.9.0? I'm in need of InOutPin on gpio. Cloning source and including turned into a lot of overhead and other issues that I'd rather not deal with if a new release is possible :)

So is it possible? Or should I try to fix the issues by working with source?

Thanks in advance!

9names, 2023-06-17:
The recommend way to deal with unreleased versions is to have a `patch.crates-io` section in your cargo.toml.
It's much easier than cloning and including.
https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#prepublishing-a-breaking-change

oOMelon, 2023-06-17:
Aha! I didn't know about that! Thanks for the information I'll try that later today

closed 2023-06-17 by oOMelon