rp-rs/rp-hal #635: Can't access any pins above pin 29 (counting from 0)

ramenspazz, 2023-06-24:
On the RP2040, I have 40 total gpio pins, but bank0 only has pins 0 through 29 in it. How can I access pins above this? I need to do this for some pimoroni goodies that use pins 30+. Sorry for the bother!

closed 2023-06-24 by ramenspazz

jannic, 2023-06-24:
Hi @ramenspazz!
I see you already closed the ticket, so I guess you found the answer yourself.
In case somebody has the same question and finds the ticket: The RP2040 only has 30 "normal" GPIO pins, the pins 0-29 in bank 0 that you already found.
In addition, there are 6 QSPI pins that are usually used for flash access. On a hypothetical device not using any flash, they could be used as additional GPIO pins. I don't know of any of-the-shelf boards using such a setup. If you know one, please tell us!

anall, 2023-06-25:
For the record: I'm not aware of any off-the-shelf boards either. I pulled the flash chip off an Adafruit Feather RP2040 [which, at the time, had the largest flash chip I could find on an off-the-shelf board] and soldered magnet wires to the pads when I was testing my GPIO code.