rp-rs/rp-hal #650: How to use interrupts?

EricLBuehler, 2023-07-16:
Hello everybody,

I am trying to use interrupts for timing. I was trying to use the builtin method as defined by the blinky.rs file, but that did not work as delay_ms is an internal function. I cannot figure out where to find the #[interrupt] attribute macro.

Additionally, if there is a better way to delay for a crate outside of rp-hal, please let me know.

Any help is greatly appreciated.

nilclass, 2023-07-16:
There are a few examples for using interrupts in the [examples directory](https://github.com/rp-rs/rp-hal/tree/main/rp2040-hal/examples). Look for those that have `irq` in their name.

One way to define an interrupt handler is to use the `interrupt` macro, like:
```
#[interrupt]
fn INTERRUPT_NAME() {
}
```
where `INTERRUPT_NAME` is one of the defined interrupts: https://docs.rs/rp2040-pac/0.4.0/rp2040_pac/enum.Interrupt.html

That's how those examples do it.

A more convenient way in my opinion though is to use RTIC (https://rtic.rs), which makes it also easy to share data between interrupt handles and other tasks. There are a few examples in the examples folder of the rp-pico board, like this one: https://github.com/rp-rs/rp-hal-boards/blob/main/boards/rp-pico/examples/pico_rtic.rs

EricLBuehler, 2023-07-17:
Thank you, but where is the interrupt macro defined? I cannot seem to find it exported anywhere.

nilclass, 2023-07-17:
It is defined in the `cortex-m-rt` package ([docs](https://docs.rs/cortex-m-rt/latest/cortex_m_rt/attr.interrupt.html)), and is re-exported from the PAC here: https://github.com/rp-rs/rp2040-pac/blob/main/src/lib.rs#L38

You can import it like
```
use rp2040_hal::pac::interrupt;
```

The `rt` and `critical-section-impl` features of rp2040-hal must be enabled for it to be available (which is also why the macro can't be found in the rp2040 docs on docs.rs :cry:).

EricLBuehler, 2023-07-17:
Thank you so much!

closed 2023-07-17 by EricLBuehler

EricLBuehler, 2023-07-19:
I agree, that would be great! Is there a way to have docs generated for all features, perhaps in different directories? Just a thought.