rp-rs/rp-hal #657: Some code assumes that a critical-section impl is available

jannic, 2023-07-24:
While `critical-section` is a non-optional dependency of `rp2040-hal`, it's not sufficient to make critical sections work: There also needs to be an implementation of critical sections, which can be provided by activating the non-default feature `critical-section-impl`.

We don't want to make it a default feature, because users might want to provide their own implementation, and disabling a default feature of a crate that might be a transitive dependency of several other crates in the dependency tree is difficult.

But if it is not enabled, and some code uses a critical section, compilation fails with an ugly error message:

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/jan/.rustup/toolchains/beta-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/jan/bin:/home/jan/.local/bin:/home/jan/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustckO5ixS/symbols.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.116l1o6y0ws3ikmf.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.1240kkeaiaeptb8m.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.1olbm9dtehzbfoiv.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.1vc85p8ilhvfb0u7.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.21ffakp0r6ft9tm8.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.2mg28lqvzv0qh0nk.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.2vsr8w0m5nchtsub.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.32yemg19tcj3me2s.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.3mpgq02128zj6oly.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.3q1goxu254lbnh2x.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.4b9hxgqas1lz440o.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.4i0a6t0pf0uquejs.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.4t03c1lkizmhxq3e.rcgu.o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.5fp7o7t4eqye4fw1.rcgu.o" "--as-needed" "-L" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps" "-L" "/home/jan/rp2040/rp-rs/rp-hal/target/debug/deps" "-L" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/build/cortex-m-1733cefd90abcc73/out" "-L" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/build/cortex-m-rt-411ac116f3d48a1d/out" "-L" "/home/jan/.rustup/toolchains/beta-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-Bstatic" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/librp2040_boot2-1dcd76b2f3b687bf.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libcortex_m_rt-7323e21c59b549c4.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/librp2040_hal-f241d53391970955.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libitertools-c0c4dd7e1a6ae53d.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libeither-5f353ddb08b04ce2.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/librand_core-80ff9e69f1776a52.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libusb_device-8affccd8628e8390.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libpio-8e8e8174efaeea39.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libnum_enum-c95269096d8017e3.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libarrayvec-d76fd001016d3893.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libfrunk-de8e896049a8c9e2.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libfrunk_core-de6212ab72d321ec.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libembedded_dma-06539f764c356729.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libstable_deref_trait-f85ab0c686f2fe7a.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libfugit-e35ab124d9d7d056.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libgcd-c667f223a468aba8.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/librp2040_pac-22d363654603e777.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libcritical_section-482151d4038dcca5.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libcortex_m-627803f227186cb6.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libembedded_hal-11c3172128a46aa2.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libvoid-e51547c64fc39745.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libnb-b1059bdd4d907b32.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libnb-33b419822c7362b3.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libvolatile_register-ffde6850c2afd136.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libvcell-a3bc3c8374a67c41.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libbare_metal-07b37e8bdc532b16.rlib" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/libpanic_halt-166dfbacac86322d.rlib" "/home/jan/.rustup/toolchains/beta-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/librustc_std_workspace_core-1c99b3969922bd66.rlib" "/home/jan/.rustup/toolchains/beta-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcore-0a902ab4b6bc6124.rlib" "/home/jan/.rustup/toolchains/beta-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcompiler_builtins-3215823e2eb3c8fe.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/home/jan/.rustup/toolchains/beta-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-o" "/home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7" "--gc-sections" "--nmagic" "-Tlink.x"
  = note: rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
          
          rust-lld: warning: section type mismatch for .got.plt
          >>> <internal>:(.got.plt): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
          
          rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
          
          rust-lld: error: undefined symbol: _critical_section_1_0_acquire
          >>> referenced by lib.rs:180 (/home/jan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.1/src/lib.rs:180)
          >>>               /home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/examples/blinky-32c74e077a2f6cd7.4b9hxgqas1lz440o.rcgu.o:(critical_section::with::h0ba4113ae293ccae)
          
          rust-lld: error: undefined symbol: _critical_section_1_0_release
          >>> referenced by lib.rs:197 (/home/jan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.1/src/lib.rs:197)
          >>>               rp2040_hal-f241d53391970955.zs3dbzwwlghqwft.rcgu.o:(core::ptr::drop_in_place$LT$critical_section..with..Guard$GT$::h13e5e4c1260f7cc9) in archive /home/jan/rp2040/rp-rs/rp-hal/target/thumbv6m-none-eabi/debug/deps/librp2040_hal-f241d53391970955.rlib
          

error: could not compile `rp2040-hal` (example "blinky") due to previous error
```

From this error message, it's not obvious how to solve the issue.

This is mitigated a bit by the fact that we do activate `critical-section-impl` from the BSP crates. (See https://github.com/rp-rs/rp-hal/pull/444 for details.)

Still, it would be nice if we could improve that error message, avoid the situation altogether, or at least describe the requirement to activate `critical-section-impl` (or provide an alternative implementation) in the docs.

ithinuel, 2023-07-24:
This could be worked around by making the example explicitly require the feature to be enabled:
https://doc.rust-lang.org/cargo/reference/cargo-targets.html#the-required-features-field

The down side is that one would still need to pass it to the command line. At least it makes the error caught earlier and leaves a much less cryptic error message.

jannic, 2023-07-24:
For the examples, that's exactly what I did in https://github.com/jannic/rp-hal/commit/3d17af0852fe944fa8ba3a16e4d457f8ed759800 (I will send a PR once the PAC update is released)

However it doesn't help users who write their own firmware based on rp2040-hal without using a BSP. They still need to remember to activate `critical-section-impl` (or some alternative).

ithinuel, 2023-07-25:
We can probably do something similar to what IIRC defmt does with a symbol in linker file that's attempted to be linked if there's something missing

jannic, 2023-07-25:
I had the following idea, which unfortunately didn't work:
- always reserve Spinlock31 for critical-section-impl, independent of the feature flag
- if the feature flag isn't set, instead of calling `critical_section::set_impl!(RpSpinlockCs);`, implement the two acquire/release symbols manually, but
  - don't call them `_critical_section_1_0_acquire` / `_critical_section_1_0_release`, but give them some distinct name
  - that way, they don't collide if some other crate registers a critical-section impl
- add something like `PROVIDE(_critical_section_1_0_acquire =_critical_section_1_0_acquire_rp2040_default )` (and the same for release) to the linker script, to make the symbol known under the canonical name
  - "The PROVIDE keyword may be used to define a symbol, such as ‘etext’, only if it is referenced but not defined."
  - This seems to be exactly what we need. Didn't try it though, because my plan broke down at an earlier point

So why didn't this work? Well, besides defining the two symbols, one also needs to tell `critical-section` about the type of the restore state, which can vary between 0 and 64 bit (`()` — `u64`). This is done by activating some feature flags on `critical-section`, and decides which return type `_critical_section_1_0_acquire` has and which parameter type `_critical_section_1_0_release` expects. If no value is selected, `critical-section` defaults to `()`, which is not what we need.

But we also can't just set the feature flag, because that may collide with a custom impl which needs a different restore state.