rp-rs/rp-hal #678: Zero-length read and write should be allowed by I2C HAL

agausmann, 2023-08-30:
Masters can probe if an address is responding by sending a "zero-length" transaction, containing only the address bits, read/write bit, and the first slave ACK.

This is a de-facto standard in I2C. It's formalized in [at least one spec](https://www.oiforum.com/wp-content/uploads/CMIS5p1_Third_Party_Spec.pdf) that I'm aware of, as "acknowledge polling":

[image]

Right now, the RP2040 I2C HAL validates that the TX and RX buffers are non-empty, which prevents such transactions:

https://github.com/rp-rs/rp-hal/blob/a12bcdae6effc8d9fe58d5de3c8edf67f23fe529/rp2040-hal/src/i2c/controller.rs#L109-L122

ithinuel, 2023-08-30:
That's the thing, the hardware does not support it.
If you need to send zero length transaction, you'll need to resort to i2c pio.

jannic, 2023-08-30:
We could do something like micropython does, and implement the zero-length transaction in software, transparent to the caller:
https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2c.c#L144-L163

ithinuel, 2023-08-30:
This would require to either make the i2c driver explicitly require a time reference or make it rely on the assumption that another time reference (at known frequency) is available somewhere.
In addition to that it'd induce some surprising latency (changing modes for the pins etc).

jannic, 2023-08-30:
Yes, it's probably a bad idea to do this implicitly like in micropython. This transparency may be nice for the caller, but it's far from zero-cost.
But it would be nice to have it as an option. Like a separate function to do zero-length transfers?

(In the end, all I'm saying is that we shouldn't just close this ticket with reference to hardware capabilities, but handle it as a valid feature request.)