rp-rs/rp-hal #682: PLL frequency out of spec

9names, 2023-09-02:
Section 2.18 of  RP2040 datasheet:
• Minimum reference frequency (FREF / REFDIV) is 5MHz
• **Oscillator frequency (FOUTVCO) must be in the range 750MHz → 1600MHz**
• Feedback divider (FBDIV) must be in the range 16 → 320
• The post dividers POSTDIV1 and POSTDIV2 must be in the range 1 → 7
• Maximum input frequency (FREF / REFDIV) is VCO frequency divided by 16, due to minimum feedback divisor
Additionally, the maximum frequencies of the chip’s clock generators (attached to FOUTPOSTDIV) must be respected. For
the system PLL this is 133MHz, and for the USB PLL, 48MHz.

pico-sdk fix PR: https://github.com/raspberrypi/pico-sdk/pull/869

embassy-rp fix commit https://github.com/embassy-rs/embassy/commit/5bbed315131991745efacbaa5c384e11f704923b

jannic, 2023-09-02:
Interesting. It looks like the datasheet has changed at some point in time. Before, it listed 400MHz as the minimum frequency.
https://web.archive.org/web/20211101082621if_/https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf#%5B%7B%22num%22%3A254%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C115%2C595.784%2Cnull%5D

Was this change announced somewhere?

jannic, 2023-09-02:
It was part of datasheet revision 1.8, 17 Jun 2022: "Increased PLL min VCO from 400MHz to 750MHz for improved
stability across operating conditions"

9names, 2025-05-14:
Resolved via #688 and #773

closed 2025-05-14 by 9names