rp-rs/rp-hal #687: External Oscillator

jonathanherbst, 2023-09-08:
Hello,

I'm trying configure the RP2040 to use a 25MHz external oscillator, instead of the 12 MHz crystal + XOSC.  I've made my own board to do this, and I've verified the external oscillator signal on an oscilloscope, however when the code configures the system clock the M0+ goes into lockup state.

```rust
static EXT_OSC_FREQ: u32 = 25_000_000;

fn setup_clocks_rosc(clocks: hal::pac::CLOCKS, rosc: hal::pac::ROSC, pll_sys: hal::pac::PLL_SYS, pll_usb: hal::pac::PLL_USB, resets: &mut hal::pac::RESETS, watchdog: &mut hal::Watchdog) -> hal::clocks::ClocksManager {
    use hal::clocks::*;
    
    watchdog.enable_tick_generation(12);
    let mut clocks = ClocksManager::new(clocks);
    let rosc = hal::rosc::RingOscillator::new(rosc).initialize();

    let pll_sys = hal::pll::setup_pll_blocking(
        pll_sys,
        EXT_OSC_FREQ.Hz(),
        hal::pll::common_configs::PLL_SYS_125MHZ,
        &mut clocks,
        resets);
    let pll_sys = pll_sys.unwrap();

    let pll_usb = hal::pll::setup_pll_blocking(
        pll_usb,
        EXT_OSC_FREQ.Hz(),
        hal::pll::PLLConfig {
            vco_freq: 1200.MHz(),
            refdiv: 1,
            post_div1: 5,
            post_div2: 5,
        },
        &mut clocks,
        resets);
    let pll_usb = pll_usb.unwrap();

    let ref_clk = clocks.reference_clock.configure_clock(&pll_usb, 12.MHz());
    ref_clk.unwrap();

    let sys_clk = clocks.system_clock.configure_clock(&pll_sys, pll_sys.get_freq());
    sys_clk.unwrap();

    let per_clk = clocks.peripheral_clock.configure_clock(&clocks.system_clock, clocks.system_clock.freq());
    per_clk.unwrap();

    clocks.gpio_output2_clock.configure_clock(&pll_sys, 1u32.MHz()).unwrap();

    clocks.usb_clock.disable();
    clocks.gpio_output0_clock.disable();
    clocks.gpio_output1_clock.disable();
    clocks.gpio_output3_clock.disable();
    clocks.adc_clock.disable();
    clocks.rtc_clock.disable();

    clocks
}
```

If I change the system clock source to `rosc` everything works and I get a 1 MHz output on gpio output 2 so I believe the sys pll is setup properly.

Any ideas on things to try would be greatly appreciated.

jannic, 2023-09-08:
Unfortunately, the necessary settings for using an external clock on the XIN pin are not well documented. There's already a ticket open about that: https://github.com/raspberrypi/pico-feedback/issues/322

Perhaps just try to disable the XOSC and see if it helps?

jonathanherbst, 2023-09-08:
Thanks for the link I will monitor that.

Tried disabling the xosc, didn't make a difference.

I did get it to work though.  I lowered the output frequency of the sys pll to 100 MHz.  I don't know why that is, I thought maybe voltage droop on vcc, but I don't see see any when setting pll output higher.

```rust
hal::pll::PLLConfig {
    vco_freq: 1600.MHz(),
    refdiv: 1,
    post_div1: 4,
    post_div2: 4,
},
```

jonathanherbst, 2023-09-08:
I'm realizing it's also possible it's clocking the SSI to fast for the flash I'm using at 125 MHz `IS25LP080D`, but I'll see what I can figure out on this later.

jannic, 2023-09-14:
If that's really the case, you could build your own boot2 loader with a reduced flash speed.

To do that, you'd need to set a larger divider in https://github.com/rp-rs/rp2040-boot2/blob/main/build.rs#L36 and then compile it following the instructions at https://github.com/rp-rs/rp2040-boot2#adding-or-changing-an-existing-bootloader

jonathanherbst, 2023-11-24:
Thanks I'll try that when I get a chance, from the datasheet my clock has up to a 5ms startup time so that is a definite possibility.  For now 100 MHz is good enough for me so I've been putting this on the back burner, you're welcome to close this if you want and I'll comment if I make some progress.