rp-rs/rp-hal #707: Linker error involving _defmt_... symbols

ggtrigg, 2023-10-25:
My project is hitting these linker errors...
```
  = note: rust-lld: error: undefined symbol: _defmt_acquire
          >>> referenced by mod.rs:55 (/home/ggt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.5/src/export/mod.rs:55)
          >>>               /home/ggt/src/rust/uld_pio/target/thumbv6m-none-eabi/debug/deps/uld_pio-fc56b4b2468a6598.1rn66jhghlxi3i0w.rcgu.o:(uld_pio::__cortex_m_rt_main::h8faa842fb243fe99)
          
          rust-lld: error: undefined symbol: _defmt_release
          >>> referenced by mod.rs:71 (/home/ggt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.5/src/export/mod.rs:71)
          >>>               /home/ggt/src/rust/uld_pio/target/thumbv6m-none-eabi/debug/deps/uld_pio-fc56b4b2468a6598.1rn66jhghlxi3i0w.rcgu.o:(uld_pio::__cortex_m_rt_main::h8faa842fb243fe99)
          
          rust-lld: error: undefined symbol: _defmt_write
          >>> referenced by mod.rs:85 (src/export/mod.rs:85)
          >>>               defmt-992e6f2160af07fb.defmt.5a321192845ed39-cgu.0.rcgu.o:(defmt::export::istr::h57442d592e9bbe30) in archive /home/ggt/src/rust/uld_pio/target/thumbv6m-none-eabi/debug/deps/libdefmt-992e6f2160af07fb.rlib
          >>> referenced by mod.rs:85 (src/export/mod.rs:85)
          >>>               defmt-992e6f2160af07fb.defmt.5a321192845ed39-cgu.0.rcgu.o:(defmt::export::integers::u32::hfd709493827c47f0) in archive /home/ggt/src/rust/uld_pio/target/thumbv6m-none-eabi/debug/deps/libdefmt-992e6f2160af07fb.rlib
```
Is this related to this crate's `critical-section-impl` feature, or something else? I couldn't find much online re these link errors.
My `Cargo.toml` dependencies look like:
```
[dependencies]
rp2040-boot2 = "0.2"
rp2040-hal = { version="0.9", features=["rt", "critical-section-impl"] }
panic-halt = "0.2"
cortex-m-rt = "0.7"
pio-proc = "0.2"
pio = "0.2"
defmt = "0.3"
```

jannic, 2023-10-25:
I guess you are missing the  `-Tdefmt.x` linker flag.
It's explained in https://defmt.ferrous-systems.com/setup. For an rp2040 project, it can be configured like in https://github.com/rp-rs/rp2040-project-template/blob/main/.cargo/config.toml

ggtrigg, 2023-10-26:
I am using the linker flag, my `config.toml` looks like:
```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# Choose a default "cargo run" tool (see README for more info)
# - `probe-rs` provides flashing and defmt via a hardware debugger, and stack unwind on panic
# - elf2uf2-rs loads firmware over USB when the rp2040 is in boot mode
runner = "probe-rs run --chip RP2040"
# runner = "elf2uf2-rs -d"

rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=--nmagic",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",

  # Code-size optimizations.
  #   trap unreachable can save a lot of space, but requires nightly compiler.
  #   uncomment the next line if you wish to enable it
  # "-Z", "trap-unreachable=no",
  "-C", "inline-threshold=5",
  "-C", "no-vectorize-loops",
]

[build]
target = "thumbv6m-none-eabi"

[env]
DEFMT_LOG = "debug"
```
and the end of the link line when doing `cargo build` is:
```
"-o" "/home/ggt/src/rust/uld_pio/target/thumbv6m-none-eabi/debug/deps/uld_pio-fc56b4b2468a6598" "--gc-sections" "--nmagic" "-Tlink.x" "-Tdefmt.x"
```
which shows the `-Tdefmt.x` flag.
It's quite puzzling!

9names, 2023-10-26:
Could you try building a clean checkout of  https://github.com/rp-rs/rp2040-project-template to see if there's something in your environment messing with all builds?

ggtrigg, 2023-10-26:
It does build, so it must be something local. I'll keep digging and comparing things with the project template.

ggtrigg, 2023-10-26:
I figured it out. The `main.rs` was missing the line:
```
use defmt_rtt as _;
```
Thanks for the help.

ggtrigg, 2023-10-26:
Resolved. Issue not really pertinent to this repository.

closed 2023-10-26 by ggtrigg

9names, 2023-10-26:
Thanks for letting us know the solution.
It's really helpful when we get bug reports like this, because others can find the answer by searching issues.
It's also a good reminder - I start with the template so often I don't really think about what is required when starting from scratch.