rp-rs/rp-hal #710: Add function to retrieve unique board ID

barafael, 2023-10-28:
Apparently, the RP2040 chip has a unique board ID somewhere fetched from the flash chip (https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_unique_id/include/pico/unique_id.h).

It would be great if this HAL made the unique ID available.

barafael, 2023-10-28:
For reference, this can be done using: https://docs.embassy.dev/embassy-rp/git/rp2040/flash/struct.Flash.html#method.blocking_unique_id

jannic, 2023-10-29:
You can do the same with `rp2040-flash`, which is a separate crate, but compatible with `rp2040-hal`: https://docs.rs/rp2040-flash/latest/rp2040_flash/flash/fn.flash_unique_id.html

Example:
https://github.com/jannic/rp2040-flash/blob/master/examples/flash_example.rs#L113-L115

barafael, 2023-10-29:
Very cool, thanks!

closed 2023-10-29 by barafael