rp-rs/rp-hal #724: ClockManager "partial move"

jnthbdn, 2023-12-02:
Hello,

I'm using the main version to put the rp2040 in deepsleep. All in all it works great, but I have a small problem.

When I create a `RealTimeClock` object I have to give it ownership over the `RtcClock` in my `ClockManager`. From then on, I can no longer pass my `ClockManager` as a parameter (`&mut`) to any function... Because the borrow checker tells me that the object is `partial move`.

How can I get round this problem?

I used the example https://github.com/rp-rs/rp-hal/blob/cfd6c128a1688186f419e3bd60ce469ddeb7fb47/rp2040-hal/examples/rtc_irq_example.rs.

Here is a simplified code example.

```rust
let mut pac = pac::Peripherals::take().unwrap();
let mut clocks = hal::clocks::ClocksManager::new(pac.CLOCKS);
// [...]
let rtc = configure_rtc(pac.RTC, clocks.rtc_clock, &mut pac.RESETS, &mut core);
// [...]

loop {
    set_next_wake_up(600);
    configure_clock_low_power(&mut clocks, &xosc, XTAL_FREQ_HZ, &mut plls);
    cortex_m::asm::wfi();
    configure_clock(&mut clocks, &xosc, XTAL_FREQ_HZ, &mut plls, &mut pac.RESETS);
}

// In configure_rtc
let mut rtc = hal::rtc::RealTimeClock::new(
    pac_rtc,
    rtc_clock,
    reset,
    rtc::DateTime {
        year: 0,
        month: 1,
        day: 1,
        day_of_week: rtc::DayOfWeek::Monday,
        hour: 0,
        minute: 0,
        second: 0,
    },
)
.unwrap();
// [...]
```

I did try to make my own `ClockManager`, but I find it a bit cumbersome and there are a lot of macros!

Can you help me ? Maybe there is something I don't understand with Rust ^^

Thanks !

jnthbdn, 2023-12-02:
I just found something very strange... Why does the `new` function of `RealTimeClock` need the ownership of rtc_clock, since in the code we're just reading the frequency (where a simple reference would be sufficient)?

https://github.com/rp-rs/rp-hal/blob/cfd6c128a1688186f419e3bd60ce469ddeb7fb47/rp2040-hal/src/rtc/mod.rs#L68

ithinuel, 2023-12-03:
Hi, 

> Why does the new function of RealTimeClock need the ownership of rtc_clock, since in the code we're just reading the frequency (where a simple reference would be sufficient)?

In some cases this is to (try to) prevent other parts of the firmware from changing the frequency/settings of the clock source.

The clock related code is admittedly not yet very polished. On the other hand, this is a fairly complex topic to address as not all solutions are really practical for the common usecase the hal is used in (eg RTIC application, low power/deepsleep, some level of [DFS](https://en.wikipedia.org/wiki/Dynamic_frequency_scaling)) while still meeting the requirement from the hardware interms of clock state, stability, and dependencies.

We welcome contributions that could help resolve (at least some of) the issues.
We currently have a few "in flight" PR up for review and testing:
- https://github.com/rp-rs/rp-hal/pull/701
- https://github.com/rp-rs/rp-hal/pull/684
- https://github.com/rp-rs/rp-hal/pull/683
- https://github.com/rp-rs/rp-hal/pull/671

jnthbdn, 2023-12-03:
I just found there is a similar idea in UsbBus...
https://github.com/rp-rs/rp-hal/blob/cfd6c128a1688186f419e3bd60ce469ddeb7fb47/rp2040-hal/src/usb.rs#L441

Once again it invalidate your ClockManager, and you lost (forever) the `usb_clock` field.

ithinuel, 2023-12-03:
The typical flow is to:
1. Setup the clock 
2. Instantiate the peripherals (rtc, usb etc)
3. don't change the clock

Taking ownership of the clock implicitly freezes the clock's settings.
An option could be that we make all peripherals take ownership of their clock source. Shared clocks, like sys_peri, could be backed by an atomic reference counter behind the scene.

If a user desires the change the clock's config, they'd need to recover those clocks (by disabling or freeing the peripherals), do what ever they want with the clock, and then reenabling or reconstructing the peripheral(s).

We could apply such pattern over all the drivers, but implementing this would be an endeavour going beyond my current time capacity.

jnthbdn, 2023-12-03:
I don't totally agree with the idea of blocking clocks when a device is instantiated. In my mind, clocks remain independent of devices, and changing a clock doesn't necessarily impact the operation of the device. It might even be possible to disable (or reduce the speed of) a device if necessary (if only temporarily, as I think that's how it's presented in the datasheet).

That said, I agree with you that devices should at least provide a `free` function to retrieve the property...

Maybe the PR should go in that way...

jannic, 2023-12-03:
From my point of view, the API should prevent easy mistakes like forgetting to initialize a clock that's needed for a peripheral, or changing a clock that's in use by a peripheral by accident, eg. due to some copy & paste mistake. That goal is covered by the current API quite well.

But then, the API should not prevent more advanced use cases where the clocks are intentionally changed while being in use. Ideally, such a facility should still only allow changes that are supported by the hardware. But if it is too difficult to distinguish between valid and invalid changes programmatically, adding more "dangerous" methods would be fine as well, as long as they are clearly recognizable as such. Eg. by giving them names like `*_unchecked()` and describing the limitations in the docs.

Currently, the only way to get around the API limitations is using `unsafe` code. That's fine for really unconventional use cases, but has the big disadvantage that it opens the door for accidentally causing undefined behavior. Some middle ground that is more lenient configuration-wise, but still provides the full soundness guarantees of safe rust, is still missing.

ithinuel, 2023-12-03:
> Exactly! You deactivate the clocks, so the USB is stopped, and when you restart your clock the USB resumes. It seems to me that this is the very use case for clock enable in the clock tree (2.15 datasheet).

It doesn't `just resume`. The usb protocol is broken. The host has lost communication with the device. A whole renumeration process needs to be resumed. Any "class level" protocol has to start again from scratch.
Some class are to some extents resilient to that (HID, CDC) but thing may go terribly wrong with things like MSD.

> There should be a system in the ClockManager that checks whether a clock is "busy". 

Indeed, what I refered to earlier is one solution to that.

> But under no circumstances should a clock be "owned" by another instance.

I don't agree with that statement though. This seems to be a strong statement without strong reasons beyond `we don't currently have a nice API`.
Note that I'm **NOT** saying that peripherals _should_ `own their reference clock`. Only that `under no circumstances` is too strong a requirement.

> I quite agree with @jannic that this seems a good compromise. [...] So we could imagine having *_unchecked functions.

Indeed, that could be a satisfying "temporary" solution until something better is proposed.
Although I like when things are consistent, not all peripherals may need the same sort of API.

jnthbdn, 2023-12-12:
Partial fix with https://github.com/rp-rs/rp-hal/pull/725