rp-rs/rp-hal #731: API ergonomics for ADC free running mode when not using FIFO

jlpettersson, 2023-12-19:
It is great that we now have ADC free running mode! I was using it in a small project, but I find that the API ergonomics could be improved.

When I want to use ADC free running mode without FIFO currently, I had to configure it and use it as follows (from the example in hal source code):

```
let mut adc_fifo: AdcFifo = adc.build_fifo().set_channel(&mut adc_pin).start();

loop {
    do_something_timing_critical();

    // read the most recent value:
    if adc_fifo.read_single() > THRESHOLD {
       led.set_high().unwrap();
    } else {
       led.set_low().unwrap();
    }
}
```

For me, it is a bit confusing to use so much "fifo"-named types and methods when I only want to use _free running mode_ **without** FIFO.

How about if we could configure _ADC free running mode_ separate from FIFO configuration? e.g. something like below:

```
let mut adc_fr: AdcFreeRunning = adc.configure_free_running().set_channel(&mut adc_pin).start();

loop {
    do_something_timing_critical();

    // read the most recent value:
    if adc_fr.read_most_recent() > THRESHOLD {
       led.set_high().unwrap();
    } else {
       led.set_low().unwrap();
    }
}
```

The benefit here is that it is a bit more explicit of what I wanted, _ADC free running mode_ but no FIFO involved.

It would also be good if I could "add" a FIFO to that _free running mode_, in some way.

jannic, 2023-12-19:
What about just renaming things?
Looking at this comment, the function might as well be named  `free_running()`:
```
    /// Start configuring free-running mode, and set up the FIFO
   [...]
    pub fn build_fifo(&mut self) -> AdcFifoBuilder<'_, u16> {
```
As it might be useful to use free-running mode without the FIFO, but not the other way around, that may actually be the better choice.

jlpettersson, 2023-12-20:
>What about just renaming things?

yes, I that could be a good alternative.

>As it might be useful to use free-running mode without the FIFO, but not the other way around, that may actually be the better choice.

Yes, that is my use case. Although the FIFO gets activated anyway, but that is a small detail.

closed 2024-02-24 by jannic