rp-rs/rp-hal #741: AdcFifo not configurable with DynPin

jannic, 2023-12-22:
To configure the channel used by `AdcFifo`, one has to call the following function on `AdcFifoBuilder`:
```
     pub fn set_channel<PIN: Channel<Adc, ID = u8>>(self, _pin: &mut PIN) -> Self {               
```

But for `DynPin`, `Channel` is defined with `ID = ()`. Therefore, `set_channel` can't be with a `DynPin` as its parameter.

ithinuel, 2023-12-23:
`DynPin` can be "_concretised_" before being used in the `AdcFifoBuilder`.

jannic, 2023-12-24:
Is it the intended way of using it?
`OneShot` has a separate implementation for that case, so `read` can be called directly on a DynPin. IMHO it would be more consistent if all functions were callable on the same set of types (as long as it's meaningful).

ithinuel, 2023-12-25:
As long as the required configuration is applied on the pin before the sampling and restored when the pin's no longer used as adc input, it's fine by me :)

jannic, 2023-12-27:
> As long as the required configuration is applied on the pin before the sampling and restored when the pin's no longer used as adc input, it's fine by me :)

The additional `impl` should still require an `AdcPin<Pin<DynPinId, F, M>>`, just with a where bound `AdcPin<Pin<DynPinId, F, M>>: Channel<Adc, ID = ()>` or similar. So the proper configuration should already be in place.

closed 2024-02-24 by jannic