rp-rs/rp-hal #750: I2C on GPIO25/26 not working

laenzlinger, 2024-01-02:
I am running a test program which shows "hello world" on an I2C OLED display, when using GPIO 8/9.
[image]

When using the same code (just changing the pins) using GPIO 24/25 it does not work at all.
[image]

I am developing on a custom board (On the Pico board, GPIO25 is connected to the on-board LED). On my custom board, these pins are connected directly to with the OLED display.

I first thought it might be some connection issue (like a solder bridge or some problem on the PCB. But running a simple blink program shows, that both outputs are working correctly.

Is there any difference between the GPIO 8/9 vs GPIO 24/25?

I am out of ideas, what else I could check.

9names, 2024-01-02:
Your issue title says GPIO25/26, but your comments say GPIO24/25 - which one is correct?

I just tested using GPIO24 and GPIO25 (no target peripheral attached).
https://gist.github.com/9names/d939875f8f1fd91b7d9295ade9d91054
The pulseview trace looks correct to me.
[image]

laenzlinger, 2024-01-02:
Thanks @9names  for your test.  The title was wrong, I have changed it.

I will try to reproduce your test on my hardware

laenzlinger, 2024-01-02:
@9names I still need to check the output with a logic analyzer. 

But just using a slow oscilloscope I recognised that after running the code https://gist.github.com/9names/d939875f8f1fd91b7d9295ade9d91054

GPIO24 => High
GIPO25 => Low

You mentioned that you were not connecting a target peripheral. Were you using an external Pull-Up resistor?

9names, 2024-01-02:
No hardware is attached to the pins besides for the logic analyzer.
Internal pullups are enabled as indicated in the code.

laenzlinger, 2024-01-03:
I don't understand what is going on here. On one hand, I can toggle the outputs when using them as logical output pins, but on the other hand, when using them as I2C pins, the CLK pin is pulled to low? Is the chip defect?

9names, 2024-01-03:
no idea. oscilloscope should show if it's just a slow rise, there should be no electrical difference between GPIO mode and i2c mode as far as i know.

jannic, 2024-01-03:
> I don't understand what is going on here. On one hand, I can toggle the outputs when using them as logical output pins, but on the other hand, when using them as I2C pins, the CLK pin is pulled to low? Is the chip defect?

When you use the pin as a logical output pin (`pin.into_push_pull_output()`), the rp2040 has a transistor connecting the pin to either 3.3V or 0V, and can drive several mA of current.

In I2C mode, the transistor is only used to drive the pin low. To create a high signal, it's only using a pull up resistor, and the internal one is quite weak: 50kΩ - 80kΩ according to the data sheet. At 3.3V, that means only about 50µA of current. So the drive strength in that direction is only 1/100th of the other one. Depending on what you connect to that pin, it's entirely possible that this is not enough to pull the pin high.

Of course, the situation is the same with GPIO 8/9. So why did it work when you used them?  Perhaps you have a borderline situation, where the pull-up strength on GPIO 25 is just slightly too small to reach a proper high level, and due to some different tolerances, it does work on GPIO 8/9 just by chance?

That said, I have seen broken GPIO pins on RP2040. It may be that the pull-up circuit is just not working on your GPIO25. Diagnosing that without measurement equipment like an oscilloscope could be difficult.

laenzlinger, 2024-01-03:
Thank you very much @jannic  and @9names  for your valuable opinion.  

I did another test which makes me think that the GPIO pin is broken.

I configured Pin 25 as pull-up input pin.  I would expect to see the statically > 3.3 V on this pin. However, it turned out to be 0V and with a very low resistance (100Ohm) measured agains GND. 

Since I can drive the output of the same pin to both HIGH and LOW, it really sounds like a broken pull up circuit on this pin.

I am sorry having wasted your valuable time, I was out of ideas and the assessment with your huge experience really helped me!

For explanation:
I am in a prototype state, where I just built a first prototype of this specific I2C functionality. Unfortunately I have selected GPIO25 which is not easily exposed on the boards I have available here.I could not easily reproduce it on my own on an other hardware.

Now I will build a 2nd sample of the prototype and hope that I can confirm that it works. I will also introduce external Pull-Up resistors to see if that helps. I already tested that with 5k resistors,  but this did not make the problematic pin work.

Feel free to close this issue if that helps you. I would definitely share my findings, which might help other users in future (I was searching for similar issues before posting)

laenzlinger, 2024-01-03:
BTW I am developing agains the 0.9.1 release, not agains main

jannic, 2024-01-04:
> I configured GPIO25 as pull-up input pin. I would expect to see the statically > 3.3 V on this pin. However, it turned out to be 0V and with a very low resistance (100Ohm) measured agains GND.

Interesting. That sounds like there is some undesired connection from the pin to GND, pulling the pin down with 100Ω. That would of course completely overwhelm the pull-up resistor (at >50kΩ) and also the 5kΩ external pull-up you tried.
But when actively driving the pin high, even at 4mA drive strength, the pin will go up to about 2.2V according to figure 171 of the datasheet. That's probably enough to be detected as a high level by most devices.

> I am sorry having wasted your valuable time, I was out of ideas and the assessment with your huge experience really helped me!

Don't worry, that's the nature of embedded development. You can't view software isolated from hardware, there are always interactions, and sometimes the hardware is just broken and it's difficult to distinguish from a software bug.

laenzlinger, 2024-01-05:
I can confirm that this was a hardware problem. After exchanging the rp2040 chip (nothing else changed), it started to work as expected. Even without external pull-up. In future versions of the PCB i will add at least to option to add an external pull up. Thanks for all your help

ithinuel, 2024-01-06:
FWIW the rp2040's datasheet recommend to have external pull-ups. From experience, I'd also do so as the internal pull-ups significantly limit the maximum frequency acheivable (due to accidental clock stretching).

laenzlinger, 2024-01-07:
thanks @jannic will do on my next version. My currently used OLED display works without but I want to keep it open for other integrations. It definitely makes sense to have the flexibility to add external pull up resistors. I should have thought about it before.