rp-rs/rp-hal #754: Input Enable now turned off on ADC capable GPIOs (Errata RP2040-E6)

thejpster, 2024-01-12:
Thanks to RP2040-E6 ([Datasheet] page 629), the RP2040 B2 ROM disables the Input Enable bit on the ADC shared pins. I don't think we currently have a mechanism to turn it back on if you want to use those pins as digital GPIOs.

[Datasheet]: https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

jannic, 2024-01-12:
We have a function to manually set the Input Enable bit: https://docs.rs/rp2040-hal/latest/rp2040_hal/gpio/struct.Pin.html#method.set_input_enable

However, should it be set automatically when configuring a pin to some other function mode?

Also, on stepping B0 and B1, the IE bit gets disabled by the SDK "as early as possible after startup".
Perhaps we should do the same? I assume this is a precautionary measure to avoid metastabilities due to undefined digital values.

thejpster, 2024-01-12:
Yes, we should probably do the same. If the user asks for DynFunction::SioXXX, or DynFunction::I2C, or DynFunction::SPI or whatever, we should set the Input Enable and Output Disable bits for the pad appropriately. At startup (which is DynFunction::Null) and if they ask for DynFunction::ADC, we should turn it off.

If they want to fiddle with the settings after picking the function, we can let them do that. But selecting a function should be *sufficient*, for any RP2040 GPIO.

PR #755 (2024-03-02): Turn off input_enable for ADC capable pins
This is meant to fix #754.

~Not yet tested at all!~
Tested using on-target-tests.