rp-rs/rp-hal #760: Should there be a default impl for hal::serial::blocking::Write?

gauteh, 2024-02-02:
Hi,

in trying to fix the defmt-serial example for rp-hal I am struggling to use the blocking serial Write trait. Ideally defmt needs to do a blocking write, but it seems that it is not implemented for rp-hal. It is not possible to accept both the blocking and non-blocking since they would potentially conflict. I don't want to depend on the non-blocking, since I then would disable more sophisticated blocking impls for other HALs. Should there be a blocking impl for the rp-hal or at least opted in to the default Write impl?

I don't know where things are in newer embedded-hal to be honest.

Regards, Gaute

jannic, 2024-02-02:
I guess you the title contains a typo and should read `embedded_hal::blocking::serial::Write`?
That's indeed not implemented. I doubt anyone plans adding it, given that it's from embedded_hal 0.2, and embedded_hal 1.0 does no longer contain this function.

But there is a function `write_full_blocking` that looks like it does exactly what you need: A blocking write to a serial port. Is there anything missing from that function?
https://docs.rs/rp2040-hal/latest/rp2040_hal/uart/struct.UartPeripheral.html#method.write_full_blocking
https://docs.rs/rp2040-hal/latest/rp2040_hal/uart/struct.Writer.html#method.write_full_blocking

gauteh, 2024-02-02:
Yes, but I can't use that one generically from defmt-serial. I'll look into whether eh1 has everything I need.

jannic, 2024-02-02:
If you want to use a trait, on the `main` branch `UartPeripheral` implements `embedded_io::Write`, which has a method `write_all`.
https://docs.rs/embedded-io/latest/embedded_io/trait.Write.html#method.write_all

`embedded_io` is more or less the official replacement for the serial functions missing from `embedded-hal` 1.0.

closed 2024-02-10 by gauteh