rp-rs/rp-hal #772: ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...

1-rafael-1, 2024-02-21:
Hi there,

this is a complete newbie question, so sorry upfront if I am missing an obvious point. I came across this repo and decided to have a shot at Rust on my Pico because lack of debuggin capability in MicroPython annoyed me. Also, why not try and learn something new? :-)

Anyway, it seems like I cannot even get to the point where I deploy an example. I followed the getting started documentation. My debug probe gets recognized:

```
❯❯❯ probe-rs list
The following debug probes were found:
[0]: Debug Probe (CMSIS-DAP) (VID: 2e8a, PID: 000c, Serial: E6633861A3543E38, CmsisDap)
````

I set up the `config.toml` to use the probe. Yet when I run the command to flash the first example to my device, the connection appears to be established but then I get the following error:

```
❯❯❯ cargo run --release --example pwm_blink
    Finished release [optimized] target(s) in 0.39s
     Running `probe-rs run --chip RP2040 target\thumbv6m-none-eabi\release\examples\pwm_blink`
      Erasing ✔ [00:00:00] [#################################################################################################] 12.00 KiB/12.00 KiB @ 63.10 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [#################################################################################################] 12.00 KiB/12.00 KiB @ 34.43 KiB/s (eta 0s )    Finished in 0.569s
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing... 
```

Now, as I am like 2 hours into fiddling my own problem solving skills are zero - and a fair amount of googling did not enlighten me at all. I am quite sure I made some very stupid mistake/omission at this point... but what? Can anybody point me into the right direction?

(side note: To cross-check I tried to flash in bootloader mode. The command runs fine. The pico is suppost to start flashing the onboard led, but it doesn't. To double-check i flashed the very same Pico with MicroPython and could confirm the device itself is operational)

ithinuel, 2024-02-21:
RTT is not used by the pwm_blink examples.
This is why the probe does not attach to it.
You can look at defmt-rtt to use with probe-rs for logging.

1-rafael-1, 2024-02-21:
thanks for the superquick reply!
Found most of the stuff, fiddled it into working. 
I can now flash via the probe, no error message. 
The led won't blink though. So maybe this is not for me... :-) There must still be some key thing I am not doing. I'll do some reading around. After all, the thing won't blink even if I flash by bootsel, so... well. 
Anyway, thanks again!

ithinuel, 2024-02-21:
Feel free to join our matrix channel and we'll be happy to help over there too :)

jannic, 2024-02-21:
> The led won't blink though

The most common cause for this is that you are using a Pico W. On that board, the LED is not directly connected to the RP2040, so it can't be controlled by toggling a GPIO pin. Instead, it's necessary to communicate with the Wifi coprocessor, which isn't implemented in rp2040-hal.

If that's the case, you have a few options:
- Use embassy-rp, which implements a wifi driver
- Connect an external LED to one of the GPIO pins and use that (don't forget a series resistor)
- Buy an RP Pico (without W)

1-rafael-1, 2024-02-22:
@jannic and @ithinuel thank You both for Your hints & tips! I managed now! (I indeed have a Pico W and indeed wiring up on e LED on another Pin did the trick)

Now I begin to know how many things I don't know :-) 

So, that being done I will next see if I can set up a new repo and slowly put all the things in place that I need to get to a flashing LED step by step in order to get an idea how this works. 
There is a ton in that small example that I do not yet understand and ... well... I'll see if I can manage. Maybe this time I need to pick up a book or tutorial first, unliked MicroPython which one can figure out (to some degree) by just reading examples.

Thanks again, this has been two fun evenings already :-)

closed 2024-02-22 by 1-rafael-1