rp-rs/rp-hal #774: Crates.io version

Foalyy, 2024-02-22:
Hi,

I've been trying to use `rp2040-hal` with `embedded-hal 1.0.0` but I'm confused by the version scheme.

Looking into the Cargo.toml in this repo, it says that this is `rp2040-hal 0.9.1` which depends on `embedded-hal 1.0.0` :
https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/Cargo.toml
So far so good.

However on crates.io, there is a 0.9.1 and a 0.9.2 with a difference source, and most notably, a different Cargo.toml that still depends on the older `embedded-hal 0.2.5` :
https://crates.io/crates/rp2040-hal/versions

Why does crates.io have both the same version number and a higher one than this repo, which actually appear to be both different (and older) than the repo itself?

I did manage to make my project work with `embedded-hal 1.0.0` by specifying this git repo instead of crates.io in my Cargo.toml so this is not a real issue for me in practice, but I'd still like to understand what I missed.

Thanks !

jannic, 2024-02-22:
Hi Foalyy!

The git repo changes with each commit, but the version number in Cargo.toml usually stays the same until we update it for the next release. So basically the only git commits where the version number is correct are the ones with a release tag. Like, for example, [v0.9.1](https://github.com/rp-rs/rp-hal/releases/tag/v0.9.1) or [v0.9.2](https://github.com/rp-rs/rp-hal/releases/tag/v0.9.2).

The situation on the main branch is double-confusing, because `Cargo.toml` still contains version 0.9.1. This is because release 0.9.2 was built from a release branch. Therefore the version bump to 0.9.2 is not visible on the main branch.

If you want to use `embedded-hal 1.0.0`, you currently have two choices: Refer to the git repo directly, as you did. Or use version 0.9.2 and activate the feature `eh1_0_alpha`. In version 0.9.2, it actually points to the released version of `embedded-hal`.

As long as you are happy with using the git version, it's probably the better choice at the moment. We'll hopefully release rp2040-hal 0.10.0 soon anyway.

Foalyy, 2024-02-22:
Hi Jannic, thanks for the quick answer and detailed explanation !

I think I understand better, I didn't think to look into the tags for the 0.9.2. During my tests I believe I tried activating `eh1_0_alpha` from the crates.io version but that didn't work, but is was probably another issue. Since the repo appeared to be explicitly compatible with `1.0.0` without a feature, I switched to a link to git.

For now, using the git repo directly is not an issue for me, and I'll switch back when `0.10.0` is released.

Thanks for your work on this repo !

closed 2024-02-22 by Foalyy