rp-rs/rp-hal #786: Does rp2040_hal::spi::Spi implement the embedded-hal 1.0 SpiDevice trait?

pdh11, 2024-03-27:
The documentation for rp2040_hal::spi::Spi starts "This struct implements the `embedded-hal` Spi-related traits."

Am I missing something here, or is that _not_ in fact the case?

I've got a W5500-EVB-Pico board -- an RP2040 microcontroller with a W5500 ethernet chip -- and I'm trying to use RP2040 SPI0 with the W5500-LL crate. The W5500 wants to deal with an embedded_hal::spi::SpiDevice (that's embedded-hal 1.0), and I'm seeing:
```
140 |         let mut w5500 = W5500::new(spi);
    |                         ---------- ^^^ the trait `embedded_hal::spi::SpiDevice` is not implemented for `Spi<Enabled, SPI0, (Pin<..., ..., ...>, ..., ...)>`

```

I mean, if I need to implement embedded_hal::spi::SpiDevice myself in terms of the rp2040 crates, then that's fine by me. But I'm loath to start into doing that when the rp2040-hal documentation plainly states that I shouldn't need to...

pdh11, 2024-03-27:
Ah, ignore that, it implements the embedded-hal-nb traits instead, that's surely better anyway.

jannic, 2024-03-27:
It also implements SpiBus, which is what HALs should do according to the embedded-hal docs: https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/index.html#for-hal-authors

To get a SpiDevice from an SpiBus, you can use https://crates.io/crates/embedded-hal-bus

pdh11, 2024-03-27:
Ah, I knew I had to be missing something, thankyou!

closed 2024-03-27 by pdh11