rp-rs/rp-hal #789: DMA chaining and looping

jsgf, 2024-04-05:
The current DMA seems a bit limited in a couple of ways.

I have an application in which I'd like to have a single buffer and transfer continuously from it to a pio fifo. When new data is ready I'd like to remove the loop, wait for DMA to quiesce and then start it on new data. I don't see any API which currently allows this.

Alternatively I'd like to double buffer, where initially its in a loop, but then I can set up a new looped buffer with new data, and repoint the old buffer to the new one, breaking its loop (with some way to know when the old buffer is no longer in use).

The current API does seem to have a way to chain buffers with `double_buffer`, but as far as I can see it only allows configuring chaining once a transfer is already underway. It doesn't let you set it up initially, for example to do scatter/gather.

I also feel like I'm missing some aspects of the current APIs. It would be helpful to have some module-level documentation to lay out a map of how its all supposed to fit together.

jsgf, 2024-04-05:
Actually I need to refresh myself on the rp2040 DMA capabilities.

jannic, 2024-04-09:
I don't think this is currently supported.

It also looks like the documentation is not correct. https://docs.rs/rp2040-hal/latest/rp2040_hal/dma/index.html contains the sentence: "This API tries to provide three types of buffers: Single buffers, double-buffered transfers where the user can specify the next buffer while the previous is being transferred, and automatic continuous ring buffers consisting of two aligned buffers being read or written alternatingly." I think this is wrong and the "continuous ring buffer" mode currently is not implemented. Or at least it's not obvious how to use the API in that mode.

Due to the way the hardware is implemented, even if you only want to send the contents of a single buffer continuously, you need two DMA channels. It is not possible to chain a channel to itself. And if the continuous mode needs two DMA channels anyway, it might be reasonable to extend `double_buffer` to support this use case. However from a usability and discoverability point of view, it might be better to have a separate mode.

jsgf, 2024-04-09:
As a side note, I was surprised that the API doesn't let you set up a chain before starting the transfer. The current API is awkward if you know there's a chain but you're essentially racing with the transfer in flight.

jsgf, 2024-04-09:
Also it doesn't have a way to group channels so you can start them simultaneously.

jannic, 2024-04-09:
> As a side note, I was surprised that the API doesn't let you set up a chain before starting the transfer. The current API is awkward if you know there's a chain but you're essentially racing with the transfer in flight.

The current API makes sense for the designated use case: In a common double-buffer setup, you want to have full access to the second buffer from your Rust code while the first buffer is used by DMA. To avoid data races (and therefore undefined behavior), the API can't setup the chaining to the second buffer before the caller cedes its reference to that buffer.

Of course, this focus on a particular use case is also limiting. If your buffers are small and the DMA speed is fast, buffer underruns become possible, and the API provides no way to compensate.

But it is difficult to design an API that is both safe and flexible. And for full flexibility, it's still possible to configure the DMA registers using the PAC.

(That doesn't mean that there's no room for improvement! And tickets like this one are useful and important to learn about actual use cases that could be better supported. I'm just explaining the status quo.)

> Also it doesn't have a way to group channels so you can start them simultaneously.

Can you provide an example where this would be important?

jsgf, 2024-04-10:
> Of course, this focus on a particular use case is also limiting. If your buffers are small and the DMA speed is fast, buffer underruns become possible, and the API provides no way to compensate.

Right. I'm thinking about things like scatter/gather cases where you've already set things up and you just want to DMA them. But you probably want to do that with a data/control pair of channels rather than dedicate one channel per buffer.

> But it is difficult to design an API that is both safe and flexible. And for full flexibility, it's still possible to configure the DMA registers using the PAC.

Yeah I'm thinking I'll fall back to that to explore the design space and then see if a reasonable safe API makes sense.

>>  Also it doesn't have a way to group channels so you can start them simultaneously.
> Can you provide an example where this would be important?

My current project is trying to drive 3 SPI DACs together in lockstep so that the samples are emitted together. The DACs themselves have a PIO SM each which are fully synchronous, but they're triggered by when the first word to output is available.

There's one DMA channel per PIO SM / DAC. Right now manually starting the DMA transfers, there's a noticeable skew across the output steams.

I'd like to start all the DMA transfers simultaneously to minimize the skew. I'm hoping that the SM FIFO buffering will be enough to cover any delayed samples due to things like DMA bus contention (as the DMA is *way* faster than the PIO clocks).