rp-rs/rp-hal #810: I2C example produces weird non-working results

HourGlss, 2024-06-14:
https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/i2c.rs

verifying with an O-Scope... the clock this produces only sends one pulse and the data line sent nothing.

ithinuel, 2024-06-16:
This has and still is being used regularly by lots of people without issues.

Could you tell us more about your use case?
- hardware setup, (simplified) schematics or off-the-shelf board reference?
- driver configuration, pin used, frequency?
- what peripheral is being controlled? have you been able to have this setup work with a different firmware?

jannic, 2024-06-16:
Just to be sure, I captured the signals on GPIO18/19 while running the i2c example. (Nothing connected to these pins, so there's no device answering.) Looks fine.
[image]

jannic, 2024-06-29:
@HourGlss any news on this? I'm quite confident that the example works as expected. Of course it is not strictly impossible that we have some bug that needs some unknown condition to get triggered. But I consider it much more likely that there's something wrong with your hardware. GPIO pins can break, for example, with the rest of the rp2040 working perfectly fine.