rp-rs/rp-hal #816: Embassy-RP and RP2040-hal Compatibility??? - Embedded_Hal::PWM:SetDutyCycle Assistance

Token-Thinker, 2024-07-15:
Hello All,

I'm building a project based on `embedded-hal` and `embassy` for various boards that does async operations. 
When doing research I noticed that the `embassy-rp` crate doesn't expose the `channel` functionality for setting the duty cycle using `embedded-hal`. 

However `rp2040-hal` does expose that feature, but doesn't seem to have any ties to embassy, which leads to my issue/question.

Will `rp2040-hal` work with `embassy` and are there any potential plans on merging the two?

9names, 2024-07-15:
> Will rp2040-hal work with embassy

No. Don't mix HALs.

> are there any potential plans on merging the two?

No. These projects are never going to merge. Feel free to copy the implementation from rp2040-hal over to embassy-rp

ithinuel, 2024-07-17:
> No. These projects are never going to merge. Feel free to copy the implementation from rp2040-hal over to embassy-rp

But be mindful of licences & a little attribution note wouldn't hurt either.

Token-Thinker, 2024-07-24:
What do you recommend than for async functionality when using rp-hal?

ithinuel, 2024-07-24:
I'd recommend, if you have the time and energy, to try to contribute the implementation of the drivers you need :)
I did that for i2c because I'm using it in a project of mine. I intend to do that for PIO at some point but I have a bunch of thing to do that are higher in priority.

closed 2024-10-29 by ithinuel