rp-rs/rp-hal #826: Add new RP2350 microcontroller support

acedogblast, 2024-08-13:
The Raspberry Pi Foundation unveiled the new RP2350 micro-controller a few days ago. A brief summary of changes and additions:
- Utilizes faster and more powerful Cortex-M33 cores (ARMv8 architecture + FPU).
- New RISC-V "hazard3" cores. Can't use ARM and RISC-V cores at the same time. Selected at the boot-loader.
- New security features. (Signed boot, encrypted firmware, SHA-256 accelerator, OTP non-volatile memory, True RNG)
- New package size (QFN-80) allows for more GPIO pins (RP235xB variants).
- Optional built-in flash 2MB (RP2354x variants).
- 520 kB of main system SRAM (near double from RP2040). 
- Enhanced and expanded PIO. 3 PIO blocks with new instructions.

Link to datasheet: https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf
RP2350 is shipping now in the Pico 2 and 3rd party development boards.

acedogblast, 2024-08-13:
Oh wow. That was fast. Thanks for your work to bring rust to the new rp chips.

9names, 2025-01-04:
We now have RP2350 support (in rp2350-hal), closing as this is resolved.

closed 2025-01-04 by 9names