rp-rs/rp-hal #843: Using sram4/5

alexkazik, 2024-08-26:
In my previous project with rust on a STM32 I managed to place assembly code in ram. The code is compiled with gcc via build.rs and cc. And is more or less this:
```
.section .data
.global mode0
.p2align 2
.type mode0,%function
mode0:
.fnstart
... code
```
and
```
  RAM  (rwx) : ORIGIN = 0x20000000, LENGTH = 31K
```

I'd like to use sram4 and 5 for the assembly code, I tried several things and it does compile but the elf says:
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
...
  4 .sram4and5    0000004c  00000000  00000000  000086e0  2**2
                  CONTENTS, READONLY
...
```

Any tips on how to do that?

Otherwise I'll try compile the assembly program with gcc into an elf, convert that to bin and include it in the rust code and copy it there myself - in the hope that I can get that to work.

thejpster, 2024-08-26:
You can also just put functions in `.data`, if you don't mind where in SRAM they go:

https://github.com/Neotron-Compute/Neotron-Pico-BIOS/blob/4bf443cd7fb35142b3f832a224ff4be93794314c/src/vga/mod.rs#L200

alexkazik, 2024-08-26:
@jannic with your linker file the resulting elf still reports that region at address 0, which will not work:
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  2 .text         00006e04  100001c0  100001c0  000002f4  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .sram4and5    0000004c  00000000  00000000  000086e0  2**2
                  CONTENTS, READONLY
```

@thejpster I do care where it is, the idea is that the second core will only run within sram4+5 to improve performance.

thejpster, 2024-08-26:
You should benchmark carefully to ensure it actually *does* improve performance, as SRAM0-3 are striped so each successive word comes from a different bank.

jannic suggested a section and a segment called `sram4` but your ELF contains a segment called `sram4and5`. Can we see the source code? If not (e.g. because this is a proprietary project) you may require professional support under NDA.

alexkazik, 2024-08-26:
sorry, I did just put the two together:
```
MEMORY {
    BOOT2 : ORIGIN = 0x10000000, LENGTH = 0x100
    FLASH : ORIGIN = 0x10000100, LENGTH = 2048K - 0x100
    RAM : ORIGIN = 0x20000000, LENGTH = 256K
    SRAM4AND5 : ORIGIN = 0x20040000, LENGTH = 8k
}

EXTERN(BOOT2_FIRMWARE)

SECTIONS {
    /* ### Boot loader */
    .boot2 ORIGIN(BOOT2) :
    {
        KEEP(*(.boot2));
    } > BOOT2
} INSERT BEFORE .text;

SECTIONS {
    .sram4and5 :
    {
	KEEP(*(.sram4and5));
    } > SRAM4AND5 AT>FLASH
} INSERT AFTER .text;
```

jannic, 2024-08-26:
> .section .sram4and5

Please try `.section .sram4and5,"ax",%progbits` here.
I can't claim I fully understand why, but it seems to be important. I guess it's the `a` flag (SHF_ALLOC, "This section occupies memory during process execution.") that makes a difference.

References:
https://developer.arm.com/documentation/101754/0618/armclang-Reference/armclang-Integrated-Assembler/Section-directives
https://www.man7.org/linux/man-pages/man5/elf.5.html

alexkazik, 2024-08-27:
That looks way better.
```
  3 .sram4and5    0000004c  20040000  10006fc4  00007118  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```
but it does not run, probably I have to copy the code? I'll continue this evening but wanted you to know now.

jannic, 2024-08-27:
Yes, that's what I meant when I wrote "some code which copies the code to RAM".
You could for example use `#[pre_init]` to copy the code from flash to RAM. But make sure to read the warnings in https://docs.rs/cortex-m-rt/0.7.3/cortex_m_rt/attr.pre_init.html, there are some pitfalls. Perhaps consider writing it in assembly.

alexkazik, 2024-08-27:
I did got it to work with:
```rust
    unsafe {
        extern "C" {
            static mut _srelocate: u8;
            static mut _erelocate: u8;
            static mut _xrelocate: u8;
        }
        let srelocate: *mut u8 = addr_of_mut!(_srelocate);
        let erelocate: *const u8 = addr_of!(_erelocate);
        let xrelocate: *const u8 = addr_of!(_xrelocate);
        let size = erelocate.offset_from(srelocate) as usize;
        from_raw_parts_mut(srelocate, size).copy_from_slice(from_raw_parts(xrelocate, size));
    }
    let _test = core1.spawn(
        unsafe {
            &mut *(slice_from_raw_parts_mut((0x20040000 + 8 * 1024 - 4 * 100) as *mut usize, 100))
        },
        move || unsafe { basic() },
    );
```

probably not the nicest but it's working. the first core uses the "normal" way and the second only this.

btw. the rp2040 datasheet says that if a ram 0-5 is accessed by more than one cpu/other than they have to wait, since sram4+5 is exclusively used by the second core no waiting should happen.

thanks for all you help!

jannic, 2024-08-27:
Glad that it finally worked!
Regarding the possible wait states on concurrent access: I'd love to see some benchmark showing how big the difference is. I guess it's rather small, but actual measurements would be interesting.

alexkazik, 2024-08-28:
I won't measure it, but just believe them that there is no wait states if not used elsewhere.

And as already said: Thanks!

closed 2025-04-22 by alexkazik