rp-rs/rp-hal #856: rpi 2350 / Pico2 - how to build working bin file ?

ds2k5, 2024-09-28:
Hi,
try to get blinky working.... but no LED is blinking....

1. git clone https://github.com/rp-rs/rp-hal/
2. rp-hal/rp235x-hal-examples/
3. cargo build --release / cargo build --release --target riscv32imac-unknown-none-elf (this one fail)
4. plug pico2 with USB to PC until pressing button
5. copy the file target/thumbv8m.main-none-eabihf/release/blinky to the mount point of pico2
6. unplug USB
7. plug USB

rustc 1.81.0 (eeb90cda1 2024-09-04)

What did I wrong ?

9names, 2024-09-29:
The thing you build with cargo is an Elf file, but you need a uf2 if you want to drag and drop it onto the drive while your Pico is in USB boot mode.
You can make a uf2 file using elf2uf2, which is part of the Pico SDK.

You can also load an elf directly with picotool from the official sdk, the examples are already configured to do that with `cargo run`:
 https://github.com/rp-rs/rp-hal/blob/c8d6e7a2d1d6a0c041cfadacc66ce812314f5108/rp235x-hal-examples/.cargo/config.toml#L37

Alternatively, you can install a fork of elf2uf2-rs via

```
cargo install elf2uf2-rs --locked --git https://github.com/thejpster/elf2uf2-rs --branch remove-arm-checks
```

and use that to make a uf2 file.

ds2k5, 2024-09-29:
@9names 

thanks

target/riscv32imac-unknown-none-elf/release$ elf2uf2-rs blinky
Error: "Not an ARM executable"

$ file blinky
blinky: ELF 32-bit LSB executable, UCB RISC-V, RVC, soft-float ABI, version 1 (SYSV), statically linked, not stripped

/target/thumbv8m.main-none-eabihf/release$ elf2uf2-rs blinky
Error: "HARD-FLOAT not supported"

target/thumbv8m.main-none-eabihf/release$ file blinky
blinky: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped

9names, 2024-09-29:
hmm, looks like elf2uf2-rs might not work with rp2350 (I thought this was tested?)

However, I did just test the following combinations, and they all worked without issue.
```
cargo run --release --bin blinky
```

```
cargo run --release --bin blinky --target  riscv32imac-unknown-none-elf
```

as well as manually invoking picotool:
```
cargo build --release --bin blinky
picotool load -u -v -x -t elf target/thumbv8m.main-none-eabihf/release/blinky
```

```
cargo build --release --bin blinky
picotool load -u -v -x -t elf target/thumbv8m.main-none-eabihf/release/blinky
```

converting to uf2 using picotool, then copying onto the mounted RP2350 boot drive worked too.

```
picotool uf2 convert target/thumbv8m.main-none-eabihf/release/blinky -t elf blinky_arm.uf2 -t uf2
picotool uf2 convert target/riscv32imac-unknown-none-elf/release/blinky -t elf blinky_rv32.uf2 -t uf2
cp blinky_arm.uf2 /media/nine/RP2350
```

ds2k5, 2024-09-29:
thanks... but picotool did not work for me...

```
$ cargo run --release --bin blinky --target  riscv32imac-unknown-none-elf
    Finished `release` profile [optimized] target(s) in 0.09s
     Running `picotool load -u -v -x -t elf target/riscv32imac-unknown-none-elf/release/blinky`
ERROR: unexpected option: -u

SYNOPSYS:
    picotool load [-v] [-x] <filename> [-t <type>] [-o <offset>] [--bus <bus>] [--address <addr>]

Use "picotool help load" for more info

```

9names, 2024-09-29:
maybe your picotool is too old?
grab a new one from https://github.com/raspberrypi/pico-sdk-tools/releases/tag/v2.0.0-5

ds2k5, 2024-09-29:
picotool uf2 convert blinky -t bin blinky.uf2
cp blinky.uf2 to RP2350

tryed with: thumbv8m.main-none-eabihf and riscv32imac-unknown-none-elf

but LED is not blinking

9names, 2024-09-29:
that's not the correct command, i gave it to you earlier.
`convert blinky -t bin` means:  blinky is a raw binary file. but it's not a raw binary, it's and elf file.
you need to use
`convert blinky -t elf`
to tell it the input is an elf file

ds2k5, 2024-09-29:
Thank you so much ! now it is working!
have a nice day!

Thanks!

closed 2024-09-29 by thejpster