rp-rs/rp-hal #875: RP2350-E9 Erratum Workaround Support

poita66, 2024-11-18:
Hi!

I can't find any information on whether rp235x-hal has any built-in workarounds for the 2V latching issue detailed in https://github.com/raspberrypi/pico-feedback/issues/401

Is this something that is needed in the HAL? Or is it irrelevant?

Thanks for reading!

jannic, 2024-11-18:
Currently there's no such support in the HAL.

I don't think it's strictly necessary to have it in the HAL, as the user can implement whatever workaround is necessary for the given hardware design. And without knowing what's connected to the pin, the HAL can't automatically decide the best way to handle it.

I may change my mind if the C SDK implements some default workaround. According to https://github.com/raspberrypi/pico-sdk/issues/1914, that's still an open issue.

Or of course if someone comes up with an elegant way to handle this in the HAL.

9names, 2025-05-14:
Immediately after jannic posted that link to the pico-sdk issue the raspberry pi folks posted this reply and then closed the issue:

> We won't be providing a software workaround in the SDK as it isn't clear what you'd do. The user can use an external resistor

The hardware solution to this problem is the best one.
I think everyone knows about this problem by now, so it's unlikely to affect anyone except those breadboarding (and those folks can easily add a resistor to their circuit)

ithinuel, 2026-11-03:
I encountered this issue while running our on-target-tests. We might need to update the Readme with some extra hw setup instruction.
I’ll report back when I have something stable.