rp-rs/rp-hal #876: PIO fails when used on core1 but works on core0

NotQuiteApex, 2024-11-18:
Howdy! I'm running into an issue where some PIO code seemingly locks up core1 (all things executing on core1 cease, and any messages made with defmt fail to make it back to my computer), whereas it works fine on core0. The code is based off the [ST7789 example from pico-examples](https://github.com/raspberrypi/pico-examples/tree/master/pio/st7789_lcd). Note that I am not using the same pins as the original example. Interestingly, I can get [ws2812-pio](https://github.com/rp-rs/ws2812-pio-rs) working on core1 just fine. I also know that the ST7789 driver from the example does work on core1 when writing C code.

I've uploaded the code here: https://github.com/NotQuiteApex/pio-core1-issue

NotQuiteApex, 2024-11-18:
That seems to have been it, thank you so much. Should've been obvious lol. Is there a better way to catch these sorts of errors so that it doesn't leave me stumped for a day and a half?

closed 2024-11-18 by NotQuiteApex

jannic, 2024-11-19:
I don't know. I guess that the HardFault handler doesn't have a chance to write out a log message because to do so it also needs some stack space. So all you get is silence. Maybe that's one of the cases where using gdb to step through the program has advantages compared to defmt logging?