rp-rs/rp-hal #883: Using one shot ADC with embedded_hal 1.0.0

laenzlinger, 2024-12-23:
I was wondering how a one-shot ADC would look like without a dependency to embedded-hal 0.2.x

I would like to implement very slow conversion (max 1Hz) on two ADC pins. So far I was using a one shot conversion which worked out well on a  rtic task  regularly triggered. (Jitter is not much of a concern for my application)

Is there a simple way to achieve this 2 input scenario without embedded-hal 0.2.x

thejpster, 2024-12-23:
It does look like the only API to do a one-shot read is through the `OneShot` trait from `embedded-hal-0.2`. Is there a reason you want to avoid a dependency on that crate?

I can think of three options:

* Use the `wait_ready`, `free_running`, `read_single` and `stop` functions.
* Set up a FIFO with any sample rate and a threshold of 1, unpause it on the RTIC timer and pause it once one sample has been read.
* Submit a PR adding an intrinsic method that does that the embedded-hal 0.2 `OneShot::read` function does.

laenzlinger, 2024-12-25:
>  Is there a reason you want to avoid a dependency on that crate

Probably no good reason. Maybe I found it a bit unusual to use two different versions of embedded_hal. But I guess that's not a good reason - it works.

I have rewritten my code to use the FIFO with round robin on the two ADC pins. This works good for my use case. Looks even a bit simpler to me  than triggering two one shot conversions individually.

Thank you so much for your support.

closed 2024-12-25 by laenzlinger