rp-rs/rp-hal #891: Support for pins in binary info

cromefire, 2025-01-30:
It seems picotool can read information about [what pin does what](https://github.com/raspberrypi/picotool?tab=readme-ov-file#pins) from the info block. Would be super useful to have that supported by the binary info crate as well and doesn't look too dissimilar from the other info [on the surface](https://github.com/raspberrypi/pico-sdk/blob/95ea6acad131124694cda1c162c52cd30e0aece0/src/common/pico_binary_info/include/pico/binary_info/code.h#L210) (not a C/C++/skilled low-level dev though).

thejpster, 2025-01-30:
Sounds like a good idea. Do you want to try and make a PR?

cromefire, 2025-01-30:
I think that's way beyond my skill level, I'm happy to get stuff working by adapting some samples... While I can see that the pins seem to be similar, I'm not sure how the address layout works there, neither in C, nor in rust.