rp-rs/rp-hal #899: RP2350 multcore enable FPU

marius-meissner, 2025-02-08:
Apparently FPU is only activated by default for the first, but not the second core.
Does it make sense to do this by default for the second core as well?

By default running any floating point operations on the second core crashes:
```Rust
let mut mc = Multicore::new(&mut pac.PSM, &mut pac.PPB, &mut sio.fifo);
let cores = mc.cores();
let core1 = &mut cores[1];

let _test = core1.spawn(unsafe { &mut CORE1_STACK.mem }, move || {
    loop {
        for i in 0..20 {
            timer.delay_ns((i as f64 * 20.33333) as u32)
        }

        onboard_led.set_high().unwrap();
    }
});
```

By enabling FPU access, the example succeeds:
```Rust
let _test = core1.spawn(unsafe { &mut CORE1_STACK.mem }, move || {
    unsafe {
        // SCB CPACR Register Address (Coprocessor Access Control Register)
        let scb_cpacr = 0xE000ED88 as *mut u32;

        // Enable CP10 and CP11 (FPU access)
        core::ptr::write_volatile(scb_cpacr, core::ptr::read_volatile(scb_cpacr) | (0xF << 20));

        asm::dsb(); // Data Synchronization Barrier
        asm::isb(); // Instruction Synchronization Barrier
    }

    loop {
        for i in 0..20 {
            timer.delay_ns((i as f64 * 20.33333) as u32)
        }

        onboard_led.set_high().unwrap();
    }
});
```

jannic, 2025-02-08:
Thanks for letting us know. This should definitely happen by default when spawning code on the second core, the current behavior is a bug.

thejpster, 2025-02-08:
yeah I can't think why you'd ever not want the FPU enabled

PR #900 (2025-02-09): Enable co-processors when spawning on core1
Fixes #899
commit f98b41e1 (2025-02-09): Enable co-processors when spawning on core1 (#900)

Fixes #899
closed 2025-02-09 by jannic