rp-rs/rp-hal #907: How to use Generics with `set_duty` from PWM

DavidFrings, 2025-03-09:
I'm trying to implement a generic `Motor` struct using `embedded-hal` and `rp2040-hal`. However, I am facing an issue when calling `set_duty_cycle_percent` (or any other `set_duty` method) on a generic PWM channel.

### Example Code:
```rust
use embedded_hal::pwm::SetDutyCycle;
use rp2040_hal::pwm::{Channel, ChannelId, FreeRunning, Slice, SliceId};

pub struct Motor<T, U, V, W>
where
    T: SliceId,
    U: SliceId,
    V: ChannelId + SetDutyCycle,
    W: ChannelId + SetDutyCycle,
{
    pub pin_0: Channel<Slice<T, FreeRunning>, V>,
    pub pin_1: Channel<Slice<U, FreeRunning>, W>,
}

impl<T, U, V, W> Motor<T, U, V, W>
where
    T: SliceId,
    U: SliceId,
    V: ChannelId + SetDutyCycle,
    W: ChannelId + SetDutyCycle,
{
    pub fn forward(self, speed: u8) {
        self.pin_0.set_duty_cycle_percent(speed).expect("PWM-Error");
        self.pin_1.set_duty_cycle_fully_off().expect("PWM-Error");
    }
}
```

### Error:
```
error[E0599]: no method named `set_duty_cycle_percent` found for struct `rp2040_hal::pwm::Channel` in the current scope
  --> src\motor.rs:91:20
   |
91 |         self.pin_0.set_duty_cycle_percent(speed).expect("PWM-Error");
   |                    ^^^^^^^^^^^^^^^^^^^^^^ method not found in `Channel<Slice<T, FreeRunning>, V>`
```

This error occurs with all variations of `set_duty`.

### Question:
How can I properly use generics to call `set_duty` methods on `Channel<Slice<T, FreeRunning>, V>`? Do I need additional trait bounds or a different approach?

Any help would be appreciated!

jannic, 2025-03-09:
You can specify that you need a Channel that implements SetDutyCycle:

```
use rp2040_hal::pwm::{Channel, FreeRunning, Slice, SliceId, ChannelId};

pub struct Motor<T, U, V, W>
where
    T: SliceId,
    U: SliceId,
    V: ChannelId,
    W: ChannelId,
    Channel<Slice<T, FreeRunning>, V>: SetDutyCycle,
    Channel<Slice<U, FreeRunning>, W>: SetDutyCycle,
{
    pub pin_0: Channel<Slice<T, FreeRunning>, V>,
    pub pin_1: Channel<Slice<U, FreeRunning>, W>,
}

impl<T, U, V, W> Motor<T, U, V, W>
where
    T: SliceId,
    U: SliceId,
    V: ChannelId,
    W: ChannelId,
    Channel<Slice<T, FreeRunning>, V>: SetDutyCycle,
    Channel<Slice<U, FreeRunning>, W>: SetDutyCycle,
{
    pub fn forward(&mut self, speed: u8) {
        self.pin_0.set_duty_cycle_percent(speed).expect("PWM-Error");
        self.pin_1.set_duty_cycle_fully_off().expect("PWM-Error");
    }
}
```

While all channels in rp2040-hal implement `SetDutyCycle`, and you can't define your own because it is `Sealed`, the compiler doesn't know that. (Or doesn't use that knowledge). So you need to specify it manually.

DavidFrings, 2025-03-09:
It work! Thank you!

closed 2025-03-09 by DavidFrings