rp-rs/rp-hal #910: rp_cargo_bin_name / rp_cargo_homepage_url not reachable from rp-binary-info

Hecatron, 2025-03-12:
Hi,
I've tried copying the rp235x-hal-examples as a basis for a new project
But what I'm noticing is that the following macro's are not visible from rp-binary-info

  * rp_cargo_bin_name
  * rp_cargo_homepage_url

As far as the rp235x-hal-examples itself that's working fine as it's referencing the directory via a path statement
But as soon as you try and reference rp235x-hal or rp-binary-info directly without that path statement, the macros are not visible for some reason

```bash
# Replace this in rp235x-hal-examples/Cargo.toml
rp235x-hal = {path = "../rp235x-hal", version = "0.3.0", features = ["binary-info", "critical-section-impl", "rt", "defmt"]}
# With this
rp235x-hal = {version = "0.3.0", features = ["binary-info", "critical-section-impl", "rt", "defmt"]}
# Then run
cargo build --bin=blinky
```

[image]

jannic, 2025-03-12:
You are right: These functions were added to rp-binary-info _after_ its 0.1.0 release.
So we should release rp-binary-info version 0.1.1 and depend on it from rp235x-hal-examples.
PR #911 prepares that release.

thejpster, 2025-04-29:
Is this now fixed?

Hecatron, 2025-04-29:
> Is this now fixed?

I think so just a case of referencing version 0.1.1 of rp-binary-info

closed 2025-04-29 by Hecatron