rp-rs/rp-hal #912: `multicore::Core::spawn`'s entrypoint should return `!` according to docs

not-nullptr, 2025-03-18:
from the docs:

> Spawn a function on this core.
> The closure should not return. It is currently defined as -> () because -> ! is not yet stable.

returning `!` has been stable for a while now -- the following code compiles under rustc 1.85.0:

```rust
fn core_entry() -> ! {
    loop {}
}
```

ithinuel, 2025-03-18:
Our current MSRV is 1.79.
I searched in the release note but couldn’t find an official statement about it.
Could you share a link to the announcement please (or the relevant tracking issue)?

not-nullptr, 2025-03-18:
https://github.com/rust-lang/rust/pull/65355 `!` was stabilised in rust 1.41.0

ithinuel, 2025-03-18:
Ah right, I see the missundestanding.

You can indeed use the `!` type in the return position of a closure or a function.
However, you still cannot define a bound to an `FnOnce() -> !`.

This causes:

```
error[E0658]: the `!` type is experimental
   --> src/multicore.rs:262:24
    |
262 |         F: FnOnce() -> ! + Send + 'static,
    |                        ^
    |
    = note: see issue #35121 <https://github.com/rust-lang/rust/issues/35121> for more information
```

not-nullptr, 2025-03-18:
what a shame... hope its stabilised soon. thanks!

closed 2025-03-18 by not-nullptr