rp-rs/rp-hal #915: Can't use spi and gpio correctly

ofluffydev, 2025-04-13:
usually repos have somewhere to ask things but this one doesn't so please answer here (no im not making an entire matrix account)

I can't get this wavershare epaper to do anything, this should be matching their clear and reset functions yet the spi and gpio logic just does nothing.

```rust
#![no_std]

use clock::XTAL_FREQ_HZ;
use embedded_hal::{
    delay::DelayNs,
    digital::{InputPin, OutputPin}, spi::SpiBus,
};
use rp235x_hal::{
    clocks::init_clocks_and_plls, gpio::{FunctionSpi, Pins}, pac::Peripherals, Clock, Sio, Spi, Timer, Watchdog
};
use embedded_time::rate::Extensions;
pub mod clock;
use crate::clock::CLOCK;

/// Enum for EPD pins
enum EpdPins {
    EpdRstPin = 12,
    EpdDcPin = 8,
    EpdBusyPin = 13,
    EpdCsPin = 9,
    EpdClkPin = 10,
    EpdMosiPin = 11,
}

// Struct for EPD
pub struct Epd2in13V4;

impl Epd2in13V4 {
    pub fn new() -> Self {
        Self
    }

    pub fn reset(&self) {
        let mut pac = Peripherals::take().unwrap();
        let sio = Sio::new(pac.SIO);
        let pins = Pins::new(
            pac.IO_BANK0,
            pac.PADS_BANK0,
            sio.gpio_bank0,
            &mut pac.RESETS,
        );

        let clocks = CLOCK.get().unwrap().lock();
        let mut timer = Timer::new_timer0(pac.TIMER0, &mut pac.RESETS, &*clocks);

        let mut epd_rst_pin = pins.gpio12.into_push_pull_output();
        epd_rst_pin.set_high().unwrap();
        timer.delay_ms(20);
        epd_rst_pin.set_low().unwrap();
        timer.delay_ms(2);
        epd_rst_pin.set_high().unwrap();
        timer.delay_ms(20);
    }

    pub fn clear(&self) {
        let width = if EPD_2IN13_V4_WIDTH % 8 == 0 {
            EPD_2IN13_V4_WIDTH / 8
        } else {
            EPD_2IN13_V4_WIDTH / 8 + 1
        };
        let height = EPD_2IN13_V4_HEIGHT;

        self.send_command(0x24); // Write Black and White image to RAM
        for _ in 0..height {
            for _ in 0..width {
                self.send_data(0xFF); // Fill with white
            }
        }

        self.turn_on_display(); // Trigger display update
    }

    fn send_command(&self, command: u8) {
        self.digital_write(EpdPins::EpdDcPin as u8, false);
        self.digital_write(EpdPins::EpdCsPin as u8, false);
        self.spi_write_byte(command);
        self.digital_write(EpdPins::EpdCsPin as u8, true);
    }

    fn send_data(&self, data: u8) {
        self.digital_write(EpdPins::EpdDcPin as u8, true);
        self.digital_write(EpdPins::EpdCsPin as u8, false);
        self.spi_write_byte(data);
        self.digital_write(EpdPins::EpdCsPin as u8, true);
    }

    fn digital_write(&self, pin: u8, value: bool) {
        let mut pac = Peripherals::take().unwrap();
        let sio = Sio::new(pac.SIO);
        let pins = Pins::new(
            pac.IO_BANK0,
            pac.PADS_BANK0,
            sio.gpio_bank0,
            &mut pac.RESETS,
        );

        match pin {
            8 => {
                let mut gpio_pin = pins.gpio8.into_push_pull_output();
                if value {
                    gpio_pin.set_high().unwrap();
                } else {
                    gpio_pin.set_low().unwrap();
                }
            }
            9 => {
                let mut gpio_pin = pins.gpio9.into_push_pull_output();
                if value {
                    gpio_pin.set_high().unwrap();
                } else {
                    gpio_pin.set_low().unwrap();
                }
            }
            10 => {
                let mut gpio_pin = pins.gpio10.into_push_pull_output();
                if value {
                    gpio_pin.set_high().unwrap();
                } else {
                    gpio_pin.set_low().unwrap();
                }
            }
            11 => {
                let mut gpio_pin = pins.gpio11.into_push_pull_output();
                if value {
                    gpio_pin.set_high().unwrap();
                } else {
                    gpio_pin.set_low().unwrap();
                }
            }
            12 => {
                let mut gpio_pin = pins.gpio12.into_push_pull_output();
                if value {
                    gpio_pin.set_high().unwrap();
                } else {
                    gpio_pin.set_low().unwrap();
                }
            }
            13 => {
                let mut gpio_pin = pins.gpio13.into_push_pull_output();
                if value {
                    gpio_pin.set_high().unwrap();
                } else {
                    gpio_pin.set_low().unwrap();
                }
            }
            _ => panic!("Invalid pin"),
        }
    }

    fn spi_write_byte(&self, byte: u8) {
        // Grab our singleton objects
        let mut pac = Peripherals::take().unwrap();

        // Set up the watchdog driver - needed by the clock setup code
        let mut watchdog = Watchdog::new(pac.WATCHDOG);

        // Configure the clocks
        let clocks = init_clocks_and_plls(
            XTAL_FREQ_HZ,
            pac.XOSC,
            pac.CLOCKS,
            pac.PLL_SYS,
            pac.PLL_USB,
            &mut pac.RESETS,
            &mut watchdog,
        )
        .unwrap();

        // The single-cycle I/O block controls our GPIO pins
        let sio = Sio::new(pac.SIO);

        // Set the pins to their default state
        let pins = Pins::new(
            pac.IO_BANK0,
            pac.PADS_BANK0,
            sio.gpio_bank0,
            &mut pac.RESETS,
        );

        // These are implicitly used by the spi driver if they are in the correct mode
        let spi_mosi = pins.gpio7.into_function::<FunctionSpi>();
        let spi_miso = pins.gpio4.into_function::<FunctionSpi>();
        let spi_sclk = pins.gpio6.into_function::<FunctionSpi>();
        let spi_bus = Spi::<_, _, _, 8>::new(pac.SPI0, (spi_mosi, spi_miso, spi_sclk));

        // Exchange the uninitialised SPI driver for an initialised one
        let mut spi_bus = spi_bus.init(
            &mut pac.RESETS,
            clocks.peripheral_clock.freq(),
            rp235x_hal::fugit::HertzU32::from_raw(16_000_000),
            embedded_hal::spi::MODE_0,
        );

        // Write the provided byte to the SPI bus
        if spi_bus.write(&[byte]).is_ok() {
            // SPI write was successful
        };

        // write 50, then check the return
        let send_success = spi_bus.write(&[50]);
        match send_success {
            Ok(_) => {
                // We succeeded, check the read value
                let mut read_buffer = [0u8; 1];
                let _ = spi_bus.read(&mut read_buffer);
                    // We got back `x` in exchange for the 0x50 we sent.
            }
            Err(_) => todo!(),
        }

        // Do a read+write at the same time. Data in `buffer` will be replaced with
        // the data read from the SPI device.
        let mut buffer: [u8; 4] = [1, 2, 3, 4];
        let mut receive_buffer: [u8; 4] = [0; 4];
        let transfer_success = spi_bus.transfer(&mut buffer, &mut receive_buffer);
        #[allow(clippy::single_match)]
        match transfer_success {
            Ok(_) => {}  // Handle success
            Err(_) => {} // handle errors
        };
    }

    fn turn_on_display(&self) {
        self.send_command(0x22); // Display Update Control
        self.send_data(0xF7); // Activate display update sequence
        self.send_command(0x20); // Execute the update
        self.wait_until_idle(); // Wait for the display to finish updating
    }

    fn wait_until_idle(&self) {
        // Wait until the busy pin goes LOW
        let mut pac = Peripherals::take().unwrap();
        let sio = Sio::new(pac.SIO);
        let pins = Pins::new(
            pac.IO_BANK0,
            pac.PADS_BANK0,
            sio.gpio_bank0,
            &mut pac.RESETS,
        );

        let mut busy_pin = pins.gpio13.into_floating_input();
        let clock = CLOCK.get().unwrap().lock();
        let mut timer = Timer::new_timer0(pac.TIMER0, &mut pac.RESETS, &*clock);
        while busy_pin.is_high().unwrap() {
            timer.delay_ms(10);
        }
    }
}

// Constants for display dimensions
const EPD_2IN13_V4_WIDTH: u16 = 122;
const EPD_2IN13_V4_HEIGHT: u16 = 250;
```

9names, 2025-04-13:
> usually repos have somewhere to ask things but this one doesn't so please answer here (no im not making an entire matrix account)

it is very strange to claim something doesn't exist, and then immediately let everyone know that you are aware that it does exist but you can't be bothered with it.
i wonder how you would respond if someone did this to you.

> I can't get this wavershare epaper to do anything, this should be matching their clear and reset functions yet the spi and gpio logic just does nothing.

the way you've written this driver it is never going to work correctly.
for one thing:
`let mut pac = Peripherals::take().unwrap();`
can only be called once, so most of this code will never be executed.

usually embedded rust driver crates should use the embedded-hal traits so that they are hal-agnostic.
as an example of that, check the `epd-waveshare` crate. it probably supports your screen anyway.

ofluffydev, 2025-04-14:
How do you correctly use the spi with this crate? The example here uses a number where a constant is expected and does not work.

https://github.com/rp-rs/rp-hal/blob/fc730fce94efdd29d570c8e14c3af62f95526fab/rp235x-hal-examples/src/bin/spi.rs#L4

ofluffydev, 2025-04-14:
this one does the same thing, you cannot just pass `_, _, _, 8` as the generics.
https://github.com/rp-rs/rp-hal/blob/fc730fce94efdd29d570c8e14c3af62f95526fab/rp235x-hal-examples/src/bin/spi_dma.rs#L2

ofluffydev, 2025-04-14:
It's also not very clear how to use thee spi when you do not use a pin for MISO, I see you can simply just not pass a `Rx` on the docs but it still doesn't explain the generics part.

closed 2025-04-14 by ofluffydev

ithinuel, 2025-04-14:
I see you closed the issue. I assume you managed to do what you want. Yet,

> usually repos have somewhere to ask things but this one doesn't so please answer here (no im not making an entire matrix account)

If you know a platform to communicate live with someone else that does not require creating an account appropriate for opensource development, please share!

From how you wrote your driver, I sense that you haven’t understood how embedded development in the rust ecosystem works and most likely have a background in scripting/memory managed languages (python/javascript?).

You are initializing the same peripherals again & again "taking" them every time, but this can only be done once. Your list of pin seem to indicate you’re using MOSI/SCLK on SPI1 but you kept the rp2350’s examples’ SPI0 on pins 4, 6 & 7.

As @9names mentionned, you don’t need to re-invent the wheel and can use [`epd-waveshare`](https://crates.io/crates/epd-waveshare).

This crate includes an example and combining the rp-hal’s example + embedded-hal doc + epd_waveshare example, here’s what you should have: https://github.com/ithinuel/sandbox-rp2040/tree/epd_waveshare_sample

Note this example uses the rp2040 but is essentially the same as rp235x for the subject matter.

ofluffydev, 2025-04-14:
I love OS dev and used rust for awhile now, just getting hit with GPIO and
micro controllers being more complex that just some pins with power. And no
I just hate matrix and there's no use trying to ask something on a platform
that's going to say every message is encrypted every time I come back. Also
GitHub discussions were literally that, not making an account since I'm
already on here.

ofluffydev, 2025-04-14:
also I don't want to use epd-waveshare I specifically want to reimplement
it myself. There only annoying issue I've ran into is `pins` not being
reusable how I'd like, is there not some easier way to store all the pins
you need without having to wrap every single one individually in a mutex?
Even if you don't use mutex, surely there's an easier way right

ofluffydev, 2025-04-14:
You're not entirely wrong tho this is a side effect of writing OOP Java for
4 years, I really just want to do get_pin() and be done most of the time.

9names, 2025-04-15:
If it's just that you don't like matrix, there are other ways to engage with the community.
`#rust-embedded` is also on IRC and all of it's history is at https://libera.irclog.whitequark.org/rust-embedded/ and is searchable - I personally use that search function a lot.
If you have Discord, there's also `#embedded` on the "Rust Programming Language Community" server.

Erhannis, 2025-05-11:
> If you know a platform to communicate live with someone else that does not require creating an account appropriate for opensource development, please share!

As OP alluded, Github permits repository owners to enable "Discussions" on repositories - seems like that would be a reasonable place for such questions.  Is it unsuitable for some reason?  (Also, I empathize with OP in that making a new account on the maintainer's favored site just to ask one question is annoying.  But I appreciate that people didn't get too riled up.)

thejpster, 2025-05-11:
I'm afraid this ticket is not the venue to ask the volunteer maintainers to open up new channels for technical support.