rp-rs/rp-hal #917: rp2040 does not compile

ofluffydev, 2025-04-14:
Versions tried:
0.9.0-0.11.0

Command used:
```
cargo run-arm --features rp2040 # And without using custom features as well, same result
```

Also tried features, did not solve issue.

Error:
```
error[E0609]: no field `rasr` on type `MPU`
  --> /home/fluffy/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/rp2040-hal-0.9.2/src/multicore.rs:72:18
   |
72 |         core.MPU.rasr.write(
   |                  ^^^^ unknown field
   |
   = note: available fields are: `_type`, `ctrl`, `rnr`, `rbar`, `rlar` ... and 7 others
```

Gustavomurta, 2025-05-07:
**RP2040 hal examples:**
https://github.com/rp-rs/rp-hal/tree/main/rp2040-hal-examples

**Getting Started**
To build all the examples, first grab a copy of the source code:

`git clone https://github.com/rp-rs/rp-hal.git`

Then use rustup to grab the Rust Standard Library for the appropriate target and our preferred flashing tool:

`rustup target add thumbv6m-none-eabi`

Then you can build the examples:

`cd rp2040-hal-examples`

`cargo build`

`cargo build --bin pio-blink` (Compile pio-blink.rs example)

`cargo install elf2uf2-rs --locked` (Install flash tool)

Boot your RP2040 into "USB Bootloader mode", typically by rebooting whilst holding "Boot Select" button

`cargo run --bin pio-blink` (Compile and flash pio-blink.rs example)

thejpster, 2025-05-07:
Yes, that's what it says in the README. I'm not sure it needed repeating here.

Gustavomurta, 2025-05-07:
The readme is not succinct and people don't like to read the whole thing. That's my impression.

ithinuel, 2025-05-08:
I’m one of those having difficulties reading long strands of text.
IMHO the readme is fine as it is. The issue here is a mix up between rp235x and rp2040.

[`run-arm`](https://github.com/rp-rs/rp-hal/blob/main/rp235x-hal-examples/.cargo/config.toml#L16) comes from the rp235x and does not exist and woun’t work for rp2040.
The information provided in the report are not enough to reproduce, let alone understand the issue.
@ofluffydev without more information we will be closing this.

jackwilsdon, 2025-05-10:
I can reproduce by building `rp2040_hal` for the wrong target (RP2350's `thumbv8m.main-none-eabihf` instead of `thumbv6m-none-eabi`) - I assume that's what's happening here. You should check your `.cargo/config.toml` or the `--target` you're passing to cargo @ofluffydev

9names, 2025-05-10:
@jackwilsdon they also did it the other way around (trying to build rp2350-hal with a target of thumbv6m-none-eabi), see https://github.com/rp-rs/rp235x-pac/issues/13

I think this question has been answered sufficiently, closing.

closed 2025-05-10 by 9names